dtcooper/raspotify #88: Service restarting endlessly and exiting with error

andy646362, 2018-02-27:
Hi everyone,

since yesterday, where I apt-get updated and apt-get dist-upgraded, Spotify will not connect to the raspotify-client. It shows up as an available device from time to time, but when I then try to connect several times it doesn't get a connection and then the device vanishes again. This is the output of sudo systemctl status raspotify:

* raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2018-02-27 21:10:56 CET; 6s ago
  Process: 2696 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --backend alsa --bitrate ${BITRATE} $CACHE_ARGS $OPTIONS (code=exited, status=1/FAILURE)
  Process: 2693 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 2690 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 2696 (code=exited, status=1/FAILURE)

Feb 27 21:10:56 osmc systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Feb 27 21:10:56 osmc systemd[1]: raspotify.service: Unit entered failed state.
Feb 27 21:10:56 osmc systemd[1]: raspotify.service: Failed with result 'exit-code'.

Which unit is entering a failed state here? I can use the USB audio just fine over airplay from iTunes.
Thanks in advance for all suggestions!

wadest, 2018-03-06:
I've got a similar problem. When system starts everything seems fine :
```
pi@jarvis:~ $ sudo systemctl status raspotify
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-03-06 14:08:43 UTC; 17s ago
  Process: 15616 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 15613 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 15619 (librespot)
   CGroup: /system.slice/raspotify.service
           └─15619 /usr/bin/librespot --name Jarvis --backend alsa --bitrate 320 --disable-audio-cache --username xxx --password xxx  --dev

Mar 06 14:08:44 jarvis librespot[15619]: INFO:librespot_core::session: Authenticated as "xxx" !
Mar 06 14:08:44 jarvis librespot[15619]: DEBUG:librespot_core::session: new Session[0]
Mar 06 14:08:44 jarvis librespot[15619]: DEBUG:librespot::spirc: new Spirc[0]
Mar 06 14:08:44 jarvis librespot[15619]: DEBUG:librespot_core::mercury: new MercuryManager
Mar 06 14:08:44 jarvis librespot[15619]: DEBUG:librespot::player: new Player[0]
Mar 06 14:08:44 jarvis librespot[15619]: INFO:librespot::audio_backend::alsa: Using alsa sink
Mar 06 14:08:44 jarvis librespot[15619]: DEBUG:librespot_core::session: Session[0] strong=3 weak=2
Mar 06 14:08:44 jarvis librespot[15619]: INFO:librespot_core::session: Country: "FR"
Mar 06 14:08:44 jarvis librespot[15619]: DEBUG:librespot_core::mercury: unknown subscription uri=hm://remote/3/user/xxx/9b8e290c8345cbc6dedf5c07d971a4
Mar 06 14:08:44 jarvis librespot[15619]: DEBUG:librespot_core::mercury: subscribed uri=hm://remote/3/user/xxx/ count=0
```
Then just after clicking on the RPi in Spotify on another device :
```
pi@jarvis:~ $ sudo systemctl status raspotify
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2018-03-06 14:09:03 UTC; 3s ago
  Process: 15619 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --backend alsa --bitrate ${BITRATE} $CACHE_ARGS $OPTIONS (code=exited, status=101)
  Process: 15616 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 15613 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 15619 (code=exited, status=101)

Mar 06 14:09:03 jarvis systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
Mar 06 14:09:03 jarvis systemd[1]: raspotify.service: Unit entered failed state.
Mar 06 14:09:03 jarvis systemd[1]: raspotify.service: Failed with result 'exit-code'.
```
Then service restarts and status is as the first shown.

matemoln, 2018-03-06:
I have commented the OPTIONS variable in the config and it solved the problem for me.

delneet, 2018-03-06:
I have a similar problem when selecting my device to play on.
```shell
mrt 06 14:49:57 pi-zero librespot[2249]: INFO:librespot_core::session: Connecting to AP "redacted"
mrt 06 14:49:58 pi-zero librespot[2249]: INFO:librespot_core::session: Authenticated as "redacted" !
mrt 06 14:49:58 pi-zero librespot[2249]: INFO:librespot_core::session: Country: "redacted"
mrt 06 14:49:58 pi-zero librespot[2249]: INFO:librespot::audio_backend::alsa: Using alsa sink
mrt 06 14:49:58 pi-zero librespot[2249]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: WireError(InvalidEnumValue(13))', /checkout/src/libcore/result.rs:860:4
mrt 06 14:49:58 pi-zero librespot[2249]: note: Run with `RUST_BACKTRACE=1` for a backtrace.
mrt 06 14:49:58 pi-zero systemd[1]: raspotify.service: main process exited, code=exited, status=101/n/a
mrt 06 14:49:58 pi-zero systemd[1]: Unit raspotify.service entered failed state.
mrt 06 14:50:08 pi-zero systemd[1]: raspotify.service holdoff time over, scheduling restart.
mrt 06 14:50:08 pi-zero systemd[1]: Stopping Raspotify...
mrt 06 14:50:08 pi-zero systemd[1]: Starting Raspotify...
mrt 06 14:50:09 pi-zero systemd[1]: Started Raspotify.
mrt 06 14:50:09 pi-zero librespot[2296]: INFO:librespot: librespot 8971d3a (2017-10-05). Built on 2017-10-05. Build ID: POyxPQVg
```
It worked just fine a couple of weeks ago.

wadest, 2018-03-06:
I just updated raspotify, and no problems anymore... I feel so dumb for not having tried it first, sorry for the inconvenience guys...

delneet, 2018-03-06:
Same thing here.
After completely removing all traces of mopidy and updating raspotify everything works again.
🤦‍♂️ 
I guess you can close this issue?

dtcooper, 2018-03-08:
Upgrade seems to have fixed this. Closing.

closed 2018-03-08 by dtcooper