dtcooper/raspotify #89: Connection timed out

zamanisaja, 2018-03-02:
started librespot with `sudo /usr/bin/librespot --cache /tmp/ -n "raspotify" --verbose --username *** --password ***`
INFO:librespot: librespot (raspotify) 8e0d758 (2018-02-26). Built on 2018-02-26. Build ID: U2Bp36G1
DEBUG:librespot_connect::discovery: Zeroconf server listening on 0.0.0.0:0
INFO:librespot_core::session: Connecting to AP "guc3-accesspoint-b-5xpp.ap.spotify.com:4070"
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Error { repr: Os { code: 110, message: "Connection timed out" } }', /checkout/src/libcore/result.rs:916:5
stack backtrace:
   0:   0x86f057 - <unknown>
   1:   0x8792bb - <unknown>
   2:   0x8678d7 - <unknown>
   3:   0x8673b3 - <unknown>
   4:   0x867d83 - <unknown>
   5:   0x867bcb - <unknown>
   6:   0x867b53 - <unknown>
   7:   0x867afb - <unknown>
   8:   0x8d0bcf - <unknown>
   9:   0x449007 - <unknown>
  10:   0x453fd3 - <unknown>
  11:   0x44c4ef - <unknown>
  12:   0x464e5b - <unknown>
  13:   0x455973 - <unknown>
  14:   0x4658e3 - <unknown>
  15:   0x867a2b - <unknown>
  16:   0x88e65b - <unknown>
  17:   0x86ece3 - <unknown>
  18:   0x456453 - main
  19: 0x76c95677 - __libc_start_main
any one else had this problem?!

ulijh, 2018-03-05:
I had the same issue, but after an upgrade to `0.10.4~librespot.20180226T171341Z.8e0d758` everything was working as normal again.

dtcooper, 2018-03-08:
Upgrade and let me know if it works, @zamanisaja

closed 2018-03-08 by dtcooper