dtcooper/raspotify #90: Non-linear volume levels

johan-andersson01, 2018-03-08:
When adjusting the volume from a client (tested on Spotify for Android and Linux), the volume increase/decrease is non-linear. 

Graphical volume bar (LEFT) vs actual approximate volume levels (RIGHT):

- 100%  : 100%
- 90% : 60%
- 75% :  20%
- 50% : 5%
- 25% : inaudible

Raspotify version: [4c41424](https://github.com/dtcooper/raspotify/commit/4c4142452fc0f090a4c1bfb1f92f5690e9a1805a)

dtcooper, 2018-03-08:
So are you saying I should remove the `--enable-volume-normalisation` flag?

johan-andersson01, 2018-03-08:
Is that the cause?

I definitely prefer normalized volumes over linear volume control, if I have to pick only one of them. But if it would be possible to have both, then that would be awesome. :smiley: 

Thanks for taking the time to develop this btw, it has really redefined my listening experience at home :heart:

dtcooper, 2018-03-09:
> Thanks for taking the time to develop this btw, it has really redefined my listening experience at home ❤️

You're welcome! I'm not sure if that's the cause, that's just the change in the commit you sent me ([`4c41424`](https://github.com/dtcooper/raspotify/commit/4c4142452fc0f090a4c1bfb1f92f5690e9a1805a)). I'll try to repro this weekend and remove it if the volume is wonky for me.

closed 2018-03-24 by dtcooper

johan-andersson01, 2018-03-25:
Works perfectly. Thanks!