dtcooper/raspotify #91: Config line in raspotify.service causes directory of "0755" to be created in root filesystem.

mikemc303, 2018-03-10:
Hello! I'm not entirely sure why this is happening, but the following line in the raspotify.service file seems to create a directory called "0755" in / whenever raspotify starts up:

`ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify ; /bin/chown raspotify:raspotify /var/cache/raspotify`

If I remove that line from the service file the directory is not created and everything works fine anyways. This is happening for me on Raspbian Stretch. Anyone else seeing this rogue directory being created? It seems to happen with every version, including the latest as of this posting.

E3V3A, 2018-03-11:
it should probably be:
```bash
ExecStartPre=/bin/mkdir -p -m 0755 /var/cache/raspotify && /bin/chown raspotify:raspotify /var/cache/raspotify
```

mikemc303, 2018-03-11:
Good catch, @E3V3A  Looks like the -m flag for mkdir is required to set the directory permissions properly.

dtcooper, 2018-03-11:
Ah, very good catch. Will fix this today.

peter739, 2018-03-11:
Great catch.  Ninja style.

closed 2018-03-13 by dtcooper

nonflammable, 2018-05-21:
issue is still present ... but in another file:
https://github.com/dtcooper/raspotify/blob/master/raspotify/DEBIAN/postinst#L15

commit 71e5911f (2018-03-13): Add -m flag to mkdir in systemd config, fixes #91.