dtcooper/raspotify #92: How to disable normalisation?

mercury44, 2018-03-10:
Hi, I'm wondering how I can disable normalisation since it is default enabled and the only argument I can give is enable-volume-normalisation?
Btw, thanks for your work!

dtcooper, 2018-03-24:
You can add whatever arguments you'd like to `VOLUME_ARGS` in `/etc/default/raspotify`. You could even start `librespot` with _no_ volumne args, taking its defaults by uncommenting the environment config variable but leaving it blank,

```
VOLUME_ARGS=
```

Closing. Let me know if that doesn't work for you.

closed 2018-03-24 by dtcooper

pjgoodall, 2020-03-25:
So I assume I can't just  make arguments up. Where do I find the list of possibles fro VOLUME_ARGS ?

dtcooper, 2020-03-25:
> So I assume I can't just make arguments up. Where do I find the list of possibles for `VOLUME_ARGS` ?
> 
> Here they are: [librespot options](https://github.com/librespot-org/librespot/wiki/Options)
> 
> Hey, and thanks! This is so good.

Looks like he answered his own question. _Pack it up, boys!_