dtcooper/raspotify #93: Additional functionality when playback starts/pauses/stops

madmodder123, 2018-03-11:
So Librespot has a --onevent option which lets you run a program before playback begins:
```
--onevent PROGRAM
        Run PROGRAM when playback is about to begin.
```
Is there a possibility we could get a similar function such as "--offevent PROGRAM" to run a program when audio has stopped playing?

I tried to get around this by running Librespot with the -v (verbose) option to grab the "command=Pause" and/or the "command=Stop" lines, but when I run it with the -v option, no audio plays. 
I am trying to get this functionality so I can start/stop my background music (python) script depending on if the user begins streaming using raspotify.

diplix, 2018-03-11:
i’d like to see the options `--onstart` `--onstop` and possibly `--onpause` to start a program, too. onstart and onstop were options that librespot supported a while ago, i think.

bvduijn, 2018-03-24:
Hi,

I created a script called setaudiodevice.sh with this contents:

```
#!/bin/bash
if [ "$PLAYER_EVENT" = "start" ]; then
  sudo amixer cset numid=3 1
elif [ "$PLAYER_EVENT" = "stop" ]; then
  sudo amixer cset numid=3 0
else
  sudo amixer cset numid=3 1
fi
```

In /etc/default/raspotify I added the options

`OPTIONS="--device hw:0 --onevent /etc/default/setaudiodevice.sh"`

This changes the audio from HDMI to audio jack when spotify start playing and on the STOP event sets it back to HDMI

Don't forget to chmod 775 on the new script!

diplix, 2018-03-26:
thanks @bvduijn, this works well for me. didn’t know that the `$PLAYER_EVENT` is beeing passed.

giggywithit, 2021-04-20:
Does the playback wait until the command is complete?

 I am looking to introduce a short delay before playback begins when spotify is started for my amp to power up. I am thinking about executing a command that will actually playback a local file comprised of silence audio file to trigger the amp, and then hopefully have spotify take over after that local tack is played..

Does this sound feasible?

microfx, 2021-11-17:
So is this working when I press pause? Stop is something in my setup that almost never is true (only when I disable Spotify Connect) - this will cause my script to never getting triggered (which is bad because it costs me money).

JasonLG1979, 2021-11-24:
@microfx this is an upstream issue. https://github.com/librespot-org/librespot Raspotify is just a packaged version of librespot.

closed 2021-11-24 by JasonLG1979