dtcooper/raspotify #94: thread main panicked at BoxAny

frederick0102, 2018-03-11:
Hi!

I listened music from my account. My girlfriend tried to connect as always, to play from her phone, and the connection lost.
I see the log file and I found this:
`INFO:librespot: librespot (raspotify) 8f8eb8c (2018-02-28). Built on 2018-03-07. Build ID: yb9b1Vkp
INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-psjr.ap.spotify.com:4070"
INFO:librespot_core::session: Authenticated as "xxxxxxxxxxxxxx" !
INFO:librespot_playback::audio_backend::alsa: Using alsa sink
INFO:librespot_core::session: Country: "HU"
INFO:librespot_playback::player: Loading track "My Amore Maria" with Spotify URI "spotify:track:4vNlVu64u9cqBGlKhLi6iu"
INFO:librespot_playback::player: Track "My Amore Maria" loaded
thread 'main' panicked at 'Box<Any>', core/src/session.rs:79:43
stack backtrace:
   0:   0x8fa63f - <unknown>
   1:   0x90489b - <unknown>
   2:   0x8f2ec7 - <unknown>
   3:   0x8f29a3 - <unknown>
   4:   0x8f3373 - <unknown>
   5:   0x604deb - <unknown>
   6:   0x67dd7f - <unknown>
   7:   0x8d461f - <unknown>
   8:   0x8d2ecf - <unknown>
   9:   0x8d7bab - <unknown>
  10:   0x4eeec3 - <unknown>
  11:   0x4dfaa3 - <unknown>
  12:   0x4efa13 - <unknown>
  13:   0x8f301b - <unknown>
  14:   0x919c33 - <unknown>
  15:   0x8fa2cb - <unknown>
  16:   0x4e0583 - main
  17: 0x76d73293 - __libc_start_main`

Raspotify:
Version: 0.10.5~librespot.20180228T144702Z.8f8eb8c

diems, 2018-03-28:
Have the same issue with Librespot. Installed a fresh Rasbian Stretch with update / upgrade

diems, 2018-03-28:
When stopping raspotify and running librespot only with the following commands, it does not crash ..
--name <..> --backend alsa --device <..> --username <..> --password <..>

diems, 2018-03-29:
It does crash unfortunately.. weird as it was going ok for a long time

closed 2021-11-24 by github-actions[bot]