dtcooper/raspotify #95: Crash on some tracks

IvanI3, 2018-03-13:
raspotify always craches on some tracks (e.g. spotify:track:0sE7yLzbBbnaSH3fCk8xgY)
and stop playing after that untill being restarted.

The logs are:

Mar 12 13:39:02 localhost librespot[904]: INFO:librespot_playback::player: Loading track "На крутом бережку (из м/ф "Приключения кота Леопольда")" with Spotify URI "spotify:track:0sE7yLzbBbnaSH3fC
k8xgY"
Mar 12 13:39:03 localhost librespot[904]: thread 'main' panicked at 'assertion failed: new_index < self.shared.chunk_count', audio/src/fetch.rs:232:9
Mar 12 13:39:03 localhost librespot[904]: stack backtrace:
Mar 12 13:39:03 localhost librespot[904]: 0: 0x54f8563f - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 1: 0x54f8f89b - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 2: 0x54f7dec7 - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 3: 0x54f7d9a3 - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 4: 0x54f7e373 - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 5: 0x54c6eff7 - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 6: 0x54c6b5cb - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 7: 0x54c6b743 - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 8: 0x54f5f61f - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 9: 0x54f5decf - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 10: 0x54f62bab - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 11: 0x54b79ec3 - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 12: 0x54b6aaa3 - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 13: 0x54b7aa13 - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 14: 0x54f7e01b - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 15: 0x54fa4c33 - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 16: 0x54f852cb - <unknown>
Mar 12 13:39:03 localhost librespot[904]: 17: 0x54b6b583 - main
Mar 12 13:39:03 localhost librespot[904]: 18: 0x76d69293 - __libc_start_main

Running latest release on RPi 3

Heil-Pepito, 2018-05-10:
Same problem here. Problem not solved since last year... (seen few threads and solutions, doesn't works)

flori4n, 2018-07-01:
same here, on a docker container on a raspberry

thread 'main' panicked at 'assertion failed: new_index < self.shared.chunk_count', audio/src/fetch.rs:232:9,
INFO:librespot_playback::player: Loading track "In Berlin" with Spotify URI "spotify:track:5aMzdlgZnDYa08Man9gm8E",
stack backtrace:,
  11:   0x90a40f - <scoped_tls::ScopedKey<T>>::set::he0659ca76ab7b331,
   3:   0x9e5f77 - std::panicking::default_hook::h637e82b93eb77513,
   0:   0x9efd1b - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::hd7376efaf737a6cb,
   1:   0x9f6d63 - std::sys_common::backtrace::print::h0b8b4ef073b17e82,
                       at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49,
                       at libstd/sys_common/backtrace.rs:71,
                       at libstd/sys_common/backtrace.rs:59,
   2:   0x9e6323 - std::panicking::default_hook::{{closure}}::h984bb18536e1dcf9,
                       at libstd/panicking.rs:207,
   4:   0x9e67cb - std::panicking::rust_panic_with_hook::h4c1dc13cc6e725f4,
                       at libstd/panicking.rs:402,
                       at libstd/panicking.rs:223,
   7:   0x5debe7 - <librespot_audio::fetch::AudioFileFetch as futures::future::Future>::poll::hd8c8d35ba13b593a,
   6:   0x5dea9f - librespot_audio::fetch::AudioFileFetch::download::h226e13f62ab0b42c,
   5:   0x5e042b - std::panicking::begin_panic::h4cf1c8cda6c092b9,
  10:   0x914347 - <tokio::executor::current_thread::scheduler::Scheduler<U>>::tick::h3b2f9eaab1a83b5f,
   8:   0x91afeb - <futures::task_impl::Spawn<T>>::poll_future_notify::hed62a615534d92ce,
   9:   0x91565f - <std::thread::local::LocalKey<T>>::with::h0d46b62ebb7297c4,
  19:   0x9e6477 - std::panicking::try::do_call::h803676a6686fa35a,
  18:   0x524cc3 - std::rt::lang_start::{{closure}}::h56d13da8f0c00177,
  17:   0x5317d3 - librespot::main::h583a074fae9ff638,
  16:   0x50e8ef - tokio_core::reactor::Core::run::h43a6a16fa99f0069,
  14:   0x9162bb - <std::thread::local::LocalKey<T>>::with::hb2e6cf69cbee52a1,
  13:   0x91643b - <std::thread::local::LocalKey<T>>::with::hd873a3d9feee8a6f,
  12:   0x915cd3 - <std::thread::local::LocalKey<T>>::with::hb116f9b6a776bfb1,
  15:   0x90ffc7 - tokio_core::reactor::Core::poll::h22adb55544452147,
                       at libstd/rt.rs:59,
                       at libstd/panicking.rs:306,
  20:   0x9fd943 - __rust_maybe_catch_panic,
                       at libpanic_unwind/lib.rs:102,
  21:   0x9e390f - std::rt::lang_start_internal::h8d13ab8f0c59613e,
                       at libstd/panic.rs:361,
                       at libstd/panicking.rs:285,
                       at libstd/rt.rs:58,
  22:   0x532223 - main,
  23: 0x76d31293 - __libc_start_main

closed 2021-11-24 by github-actions[bot]