dtcooper/raspotify #97: Spotify connect client not always visible

cslekniws, 2018-03-15:
Installed the spotify connect on my Pi2. I use HifiBerry with DAC+. OS is Raspbian (standard Raspbian lite)
On my Android phone installed Spotify. 
I can see the Spotify Connect Client and stream music. But every now and then the Connect Client disappears. Sometimes waiting a while helps. Restart the Pi does not help. 
My Pi has a fixed IP. If the Spotify Connect client is not visible i can still access my PI with Putty. No problem. 
Is there a solution for this problem?
Thanx in advance

Cheers. 
Cees

dtcooper, 2018-03-15:
Is either (a) your specified username and password in `/etc/default/raspotify` or (b) the Android phone 100% connected to the same wifi network and can ping the Pi?

cslekniws, 2018-03-15:
Added username and password to /etc/default/raspotify and did a restart.
I thought that username and password where only used to access the spotify connect via Internet and not local LAN?
Absolutely sure that the network is the same and i can ping the PI.

I will test my PI with username and password!
Thanx for the quick reply!

dtcooper, 2018-03-15:
It is only necessary for that, but your Android device hopping on to a different network could cause your problem. If you add your username/password, this should be avoided.

cslekniws, 2018-03-16:
Today the Raspotify disapeared again on the visible divices on my Spotify APP. Sometimes if i leave it playing long enough the Raspotify shows up again. 
For username and password; is it inlcuding the "<" and ">" ?
Do you know what causes the broadcast effect on the Spotify Connect to the client devices?
Maybe the ChromeCast from Google is disturbing the broadcast effect?
My PI is connected via UTP cable. Not Wifi.

dtcooper, 2018-03-16:
Yes, the number sign ("hashtag") would need to be uncomments, and the chevrons (`<` and `>`) would need to be replaced, ie

```
#OPTIONS="--username <USERNAME> --password <PASSWORD>"
```

Turns into,

```
OPTIONS="--username mygreatusername --password mygreatpassword"
```

cslekniws, 2018-03-17:
Thanx very much!!

I think this is the defenitive solution for me.

Changed the username and password Line without the chevrons (didn't know
the English word for it) and immediatly the cliënt popped up in my client
device!

Thanx again!

dtcooper, 2018-03-17:
Glad it worked out for you!

closed 2018-03-17 by dtcooper

DavidM42, 2020-01-27:
Just a note if anyone is facing this problem. If your username or password contain some special characters like semicolons and spaces you have to wrap the username or password in the config line in ' for it to work since these get passed into the terminal.
So like `OPTIONS="--username 'mygreatusername' --password 'mygreatpassword !security;' "`