dtcooper/raspotify #98: No sound first few seconds on new song?

hargibii, 2018-03-18:
As the title says. I dont get any sound for the first few seconds when changeing song.

Running a RPI3 with debian. Output is trough HDMI to an denon recever.

running the latest build from github.

aalvarez737, 2018-03-23:
I'm getting the same issue, but it seems to be related to track loading. The sound starts when the log say track loaded (systemctl status).

gabmartini, 2018-03-25:
Same here. Using Raspian Stretch and moodeAudio.

hargibii, 2018-04-11:
any way to minimize this issue?

MelBourbon, 2018-05-08:
I think as long as librespot does not support gapless playback ( https://github.com/librespot-org/librespot/issues/21) this can not be resolved since the next song needs to be preloaded.

kennell, 2018-12-28:
Same issue here after recently reinstallting everything from scratch on a clean Raspbian installation. Did not have this issue previously.

closed 2021-11-24 by github-actions[bot]