dtcooper/raspotify #99: Librespot packet error

scetu, 2018-03-31:
I'm facing bug when my Raspotify plays and after a while stops. And everytime is the same warning after running `systemctl status raspotify.service` 
I see only this warning - `Apr 01 00:34:18 raspberrypi librespot[7219]: WARN:mdns::fsm: error sending packet Error { repr: Os { code: 99, message: "Cannot assign requested address" } }` 

Someone who knows something about this bug?

Wqrld, 2018-04-08:
i also have this, its a librespot issue but i cant seem to find out why it happens

cloudsoh, 2018-08-29:
Currently facing this problem too.

tbolis, 2018-10-13:
hello same issue here, do we have any update?

```
Oct 13 17:04:34 raspberrypi librespot[8208]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign r...ddress" }
```
it plays a couple of songs and then sound seems to stop while the spotify-connect device seems to play normally

pls check

jungkarol, 2019-01-22:
Hey, the same issue in my case. Could you take a look on this?

`sty 22 19:11:39 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:12:06 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:12:55 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:12:59 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:13:49 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:14:39 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:15:28 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:16:18 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:17:07 raspberrypi librespot[13152]: INFO:librespot_playback::player: Loading track "Daddy Cool" with Spotify URI "spotify:track:3WMbD1OyfKuwWDWMNbPQ4g"
sty 22 19:17:08 raspberrypi librespot[13152]: INFO:librespot_playback::player: Track "Daddy Cool" loaded
sty 22 19:17:09 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:17:58 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:18:48 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:19:38 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:20:28 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:20:37 raspberrypi librespot[13152]: INFO:librespot_playback::player: Loading track "Simple Man" with Spotify URI "spotify:track:6F6DxQ3XjToEHX1e2KKSbg"
sty 22 19:20:39 raspberrypi librespot[13152]: INFO:librespot_playback::player: Track "Simple Man" loaded
sty 22 19:21:18 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:22:08 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:22:58 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:23:48 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:24:38 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:25:29 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
sty 22 19:26:19 raspberrypi librespot[13152]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }`

patrickjane, 2019-02-19:
Same errors for me. I have just installed raspotify for the first time, on a raspberry pi 3 with raspbian stretch installed on it. As I found in the comments, there is no stretch version, so I installed the jessi version, as other user reported it should work.

After editing the configuration & restarting, I found this in the logs:

```
Feb 19 21:13:37 pi3HA librespot[6531]: INFO:librespot: librespot (raspotify v0.11.6) d2cadec (2018-11-19). Built on 2018-11-29. Build ID: Uwq1tCGm
Feb 19 21:13:37 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:13:37 pi3HA librespot[6531]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-wfqt.ap.spotify.com:4070"
Feb 19 21:13:37 pi3HA librespot[6531]: INFO:librespot_core::session: Authenticated as "myuser" !
Feb 19 21:13:37 pi3HA librespot[6531]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
Feb 19 21:13:37 pi3HA librespot[6531]: INFO:librespot_core::session: Country: "DE"
```
Yet it started to play music, however more errors appeared:

```
Feb 19 21:14:09 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:14:12 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:14:13 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:14:14 pi3HA librespot[6531]: INFO:librespot_playback::player: Loading track "Against the Grain" with Spotify URI "spotify:track:54maxwHjnsyjWF2k3YoIE0"
Feb 19 21:14:15 pi3HA librespot[6531]: INFO:librespot_playback::player: Track "Against the Grain" loaded
Feb 19 21:14:16 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:14:25 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:14:45 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:16:24 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:16:28 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:16:29 pi3HA librespot[6531]: INFO:librespot_playback::player: Loading track "Against the Grain" with Spotify URI "spotify:track:54maxwHjnsyjWF2k3YoIE0"
Feb 19 21:16:30 pi3HA librespot[6531]: INFO:librespot_playback::player: Track "Against the Grain" loaded
Feb 19 21:16:31 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
```

At some point it gut stuck, however I don't know how I achieved it:

```
Feb 19 21:16:40 pi3HA librespot[6531]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Feb 19 21:16:41 pi3HA librespot[6531]: INFO:librespot_playback::player: Loading track "Dial My Number (feat. Kumar Bent)" with Spotify URI "spotify:track:6ybHTBbiCxjUpVQbiRg46P"
Feb 19 21:16:57 pi3HA librespot[6531]: ERROR:librespot_core::channel: channel error: 2 1
Feb 19 21:16:57 pi3HA librespot[6531]: WARN:librespot_audio::fetch: error from channel
```

After this it would not play any more music until I restarted the service.

Is there any news on this bug? I see it was reported almost a year ago already, I hope there is a way to resolve it.

closed 2021-11-24 by github-actions[bot]