dtcooper/raspotify #100: Device disappears after router restart

Zachar2, 2018-04-05:
As the title says raspotify disappears from the list of available devices in Spotify app (web player too) after restarting my router (while the service on my Raspberry Pi 3 is still running correctly). It's coming back on the list after manually restarting raspotify service in Raspbian. Any idea what might be causing this? My network has pretty basic setup and Raspberry has a static IP address. I tested this on wifi and cable with the same results.

bermudaparty, 2018-04-12:
I also had the problem of it not showing up on my Android Spotify app. What worked for me was to disable the option "Show local devices only" (even though it is a local device; see here https://community.spotify.com/t5/Newcomers-and-Contribution/Spotify-connect-on-Raspberry-Pi-HifiBerry-sometimes-not-showing/td-p/4406571).

Maybe your issue could be related to this? Somehow depending on the state of the network it doesn't get recognized as a local device..?

Zachar2, 2018-04-17:
Thank you for the advice. It looks like the problem is on the Raspbian/raspotify level because all other devices show up on the list without issues. The mystery remains unsolved.

closed 2021-11-24 by github-actions[bot]