dtcooper/raspotify #102: Using /var/cache/raspotify for local file streaming?

MatanyaLoewenthal, 2018-04-11:
Hello, I want to be able to stream local files to my RPI, so I enabled the cache part of the config file, and I'm not sure what else to do. Thanks!

cortegedusage, 2018-04-11:
unfortunately the librespot project (on which raspotify is based on) doesn't support the playing of local files, 
You could ask at the librespot org github, if you wan't it to support it.

Good luck and happy listening!

JasonLG1979, 2022-06-21:
> Did this ever get fixed? I'm logged into both the "controlling device" and raspotify as myself, so should have access to my own playlist. This works when I play my local files on mobile, but I guesss only because mobile allows you to "download" the playlist.
> 
> `Jun 21 08:37:21 hestia.winterpalace.org librespot[27170]: [2022-06-21T00:37:21Z DEBUG librespot_connect::spirc] State: context_uri: "spotify:playlist:1I2kAizki8fosJjijLQnqh" index: 4294967295 position_ms: 0 status: kPlayStatusStop position_measured_at: 1655771841731 context_description: "Testing" shuffle: true repeat: false playing_from_fallback: true` `Jun 21 08:37:21 hestia.winterpalace.org librespot[27170]: [2022-06-21T00:37:21Z DEBUG librespot_connect::spirc] Frame has 0 tracks` `Jun 21 08:37:21 hestia.winterpalace.org librespot[27170]: [2022-06-21T00:37:21Z INFO librespot_connect::spirc] Fetching autoplay context uri` `Jun 21 08:37:21 hestia.winterpalace.org librespot[27170]: [2022-06-21T00:37:21Z INFO librespot_connect::spirc] No more tracks left in queue` `Jun 21 08:37:21 hestia.winterpalace.org librespot[27170]: [2022-06-21T00:37:21Z TRACE librespot_connect::spirc] Sending status to server: [kPlayStatusStop]` `Jun 21 08:37:21 hestia.winterpalace.org librespot[27170]: [2022-06-21T00:37:21Z DEBUG librespot_playback::player] command=SetAutoNormaliseAsAlbum(false)` `Jun 21 08:37:21 hestia.winterpalace.org librespot[27170]: [2022-06-21T00:37:21Z DEBUG librespot_playback::player] command=Stop` `Jun 21 08:37:21 hestia.winterpalace.org librespot[27170]: [2022-06-21T00:37:21Z INFO librespot_connect::spirc] Autoplay uri resolved to <"spotify:station:playlist:1I2kAizki8fosJjijLQnqh">`

Do any chromecast or smart speaker like Spotify Connect devices support that feature? I don't really see it as a good fit for librespot. IMHO that's the  niche that librespot is trying to fill.

JasonLG1979, 2022-06-21:
If you want to stream local content there are many solutions for that.