dtcooper/raspotify #103: Huge delays before playing

totomaze, 2018-04-13:
Hi,
When I start a song through raspotify the sound will come only after the song is loaded.(log)
This result in huge delay before the song start playing from 30s to 1min after.

Thanks

neiljohari, 2018-04-14:
I am currently experiencing this too on a Raspberry Pi 3. I've tried playing with the cache and bitrate quality so far—no luck.

audifaxdev, 2018-04-19:
The delays varies, The song is displayed as playing for 5 to 30 seconds before I can hear the beginning of the song.

totomaze, 2018-04-20:
Update :
After my download speed came back up the delays are down to 3-10s.
I ended up using media server (windows) with the spotify app.
Thats a shame we can't run open.spotify.com on a rasberryPi
ps: I am a premium user

neiljohari, 2018-04-20:
@audifaxdev I have a premium account, still experience delays. Going to see if a new router makes any difference soon as the old one had some problems and was severely hurting our speeds.

8pointagenda, 2018-08-02:
Has anybody found a solution to this issue yet?

redcrayonn, 2018-09-21:
Issue happens continuosly for me.

charlesmillet, 2018-09-25:
I also have between 5 and 10 seconds of delay on my Pi 3 and no delay on my desktop.

thebeater, 2018-10-24:
Same here, i have an delay of minimum 8 seconds after my app is showing the playback signal.

rulingspear, 2018-12-23:
Me too. Running on a raspi 1 B+. No network issues as far as I'm concerned, other wifi enabled spotify players are instantly playing.

charlesmillet, 2019-01-25:
The delay has disappear since my internet speed has increased (I had 2MBits and now 1000).

luke-herm, 2020-03-13:
found a cheap workaround for me... playing a long, empty mp3 while raspotify is running gave me continuous playback somehow

closed 2021-11-24 by github-actions[bot]