dtcooper/raspotify #105: Random crasses

mannte, 2018-04-18:
I get random crashes. I have not found any pattern in it yet.
It stops with the following error:
`Apr 18 21:28:28 mantepi librespot[1517]: thread 'main' panicked at 'Box<Any>', core/src/session.rs:79:43
Apr 18 21:28:28 mantepi librespot[1517]: stack backtrace:
Apr 18 21:28:28 mantepi librespot[1517]:    0:   0x85180f - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:    1:   0x85ba6b - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:    2:   0x84a097 - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:    3:   0x849b73 - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:    4:   0x84a543 - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:    5:   0x55bfbb - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:    6:   0x5d4f4f - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:    7:   0x82b7ef - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:    8:   0x82a09f - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:    9:   0x82ed7b - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:   10:   0x445f8f - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:   11:   0x436b5b - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:   12:   0x446adf - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:   13:   0x84a1eb - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:   14:   0x870e03 - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:   15:   0x85149b - <unknown>
Apr 18 21:28:28 mantepi librespot[1517]:   16:   0x43764f - main
Apr 18 21:28:28 mantepi librespot[1517]:   17: 0x76d69677 - __libc_start_main
`

mannte, 2018-04-30:
Oke thx, I updated to 0.11.1, will see if it helps although I haven't got the same error yet after connecting a Ethernet cable but it was also not completely stable.

mannte, 2018-05-04:
Got a random crash again today.

Errors in syslog:
```
May  4 17:56:39 mantepi librespot[8593]: INFO:librespot_playback::player: Loading track "Are You Ready" with Spotify URI "spotify:track:5kzbtG4pIcu2s0SVrogm4o"
May  4 17:56:41 mantepi librespot[8593]: ERROR:librespot_core::channel: channel error: 2 1
May  4 17:56:41 mantepi librespot[8593]: thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: ChannelError', libcore/result.rs:945:5
May  4 17:56:41 mantepi librespot[8593]: stack backtrace:
May  4 17:56:41 mantepi librespot[8593]:    0:   0xa05333 - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h4ef6490c84cef3d0
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
May  4 17:56:41 mantepi librespot[8593]:    1:   0x9ff08f - std::sys_common::backtrace::_print::hea150ee4f33bb5de
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/sys_common/backtrace.rs:71
May  4 17:56:41 mantepi librespot[8593]:    2:   0xa03da3 - std::panicking::default_hook::{{closure}}::h6c4396ff5bd61d89
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/sys_common/backtrace.rs:59
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/panicking.rs:380
May  4 17:56:41 mantepi librespot[8593]:    3:   0xa03987 - std::panicking::default_hook::h0f6cb16cf06a0198
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/panicking.rs:396
May  4 17:56:41 mantepi librespot[8593]:    4:   0xa04247 - std::panicking::rust_panic_with_hook::h2207195920ffa55f
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/panicking.rs:576
May  4 17:56:41 mantepi librespot[8593]:    5:   0xa040ab - std::panicking::begin_panic::h8d43a09cf6d6e5f1
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/panicking.rs:537
May  4 17:56:41 mantepi librespot[8593]:    6:   0xa0402f - std::panicking::begin_panic_fmt::hd9afc94159ee9735
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/panicking.rs:521
May  4 17:56:41 mantepi librespot[8593]:    7:   0xa03fc7 - rust_begin_unwind
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/panicking.rs:497
May  4 17:56:41 mantepi librespot[8593]:    8:   0xa51e57 - core::panicking::panic_fmt::h0bd854df201d1baf
May  4 17:56:41 mantepi librespot[8593]:                        at libcore/panicking.rs:71
May  4 17:56:41 mantepi librespot[8593]:    9:   0x5b090f - core::result::unwrap_failed::h0e273f2ef825e9af
May  4 17:56:41 mantepi librespot[8593]:   10:   0x5ab85f - librespot_playback::player::PlayerInternal::run::h296c59bd7682ffbf
May  4 17:56:41 mantepi librespot[8593]:   11:   0x52b4bb - std::sys_common::backtrace::__rust_begin_short_backtrace::h0ef34e9728d818f7
May  4 17:56:41 mantepi librespot[8593]:   12:   0x53ea03 - std::panicking::try::do_call::h55970923d54b3006
May  4 17:56:41 mantepi librespot[8593]:   13:   0xa13673 - __rust_maybe_catch_panic
May  4 17:56:41 mantepi librespot[8593]:                        at libpanic_unwind/lib.rs:102
May  4 17:56:41 mantepi librespot[8593]:   14:   0x54092b - <F as alloc::boxed::FnBox<A>>::call_box::h6c97a59281f0cee4
May  4 17:56:41 mantepi librespot[8593]:   15:   0xa0a2df - std::sys::unix::thread::Thread::new::thread_start::hb1d1727b173caaf8
May  4 17:56:41 mantepi librespot[8593]:                        at /checkout/src/liballoc/boxed.rs:798
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/sys_common/thread.rs:24
May  4 17:56:41 mantepi librespot[8593]:                        at libstd/sys/unix/thread.rs:90
```

Status of the service:
```
May 04 17:56:41 mantepi librespot[8593]:   10:   0x5ab85f - librespot_playback::player::PlayerInternal::run::h296c59bd7682ffbf
May 04 17:56:41 mantepi librespot[8593]:   11:   0x52b4bb - std::sys_common::backtrace::__rust_begin_short_backtrace::h0ef34e9728d818f7
May 04 17:56:41 mantepi librespot[8593]:   12:   0x53ea03 - std::panicking::try::do_call::h55970923d54b3006
May 04 17:56:41 mantepi librespot[8593]:   13:   0xa13673 - __rust_maybe_catch_panic
May 04 17:56:41 mantepi librespot[8593]:                        at libpanic_unwind/lib.rs:102
May 04 17:56:41 mantepi librespot[8593]:   14:   0x54092b - <F as alloc::boxed::FnBox<A>>::call_box::h6c97a59281f0cee4
May 04 17:56:41 mantepi librespot[8593]:   15:   0xa0a2df - std::sys::unix::thread::Thread::new::thread_start::hb1d1727b173caaf8
May 04 17:56:41 mantepi librespot[8593]:                        at /checkout/src/liballoc/boxed.rs:798
May 04 17:56:41 mantepi librespot[8593]:                        at libstd/sys_common/thread.rs:24
May 04 17:56:41 mantepi librespot[8593]:                        at libstd/sys/unix/thread.rs:90
```

berrywhite96, 2018-05-07:
Can confirm this issue, same exception on the same line:
`thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: ChannelError', libcore/result.rs:945:5`
This crash happend after 5 songs.