dtcooper/raspotify #106: Need to restart raspotify every time I want to use it

witnessmenow, 2018-04-27:
Hello,

I set up Raspotify on an original Pi Zero (pre camera connector even) using a fresh install of Rasbian Lite (2018-04-18)

I'm using A USB hub to connect a WiFi adapter and USB sound card. USB sound card is setup as the default Audio device as descrbied in this issue https://github.com/dtcooper/raspotify/issues/40

Raspotify is working fine for me when it works, I'm just using local streaming (so not providing any spotify credentials to raspotify). 

I only set it up last night but both times I went to use it, it wasn't showing up as an connect device in spotify til I did a restart `sudo systemctl restart raspotify`

After restarting it seems to work great, I haven't rested exhaustively but I can definitely listen to 20 minutes of music uninterrupted (The only issue I have is the sound is a bit crackly for the first 10 seconds or so, but I'm not overly concerned about this as the moment)

Is there any more info that I can provide that would be helpful to debug this? The network is still definitely up as I am doing the restart via SSH

Ptafke, 2018-05-27:
Hi, I on version 0.11.2 and I have exact the same behavior.
Is there anything I can do?

irovder, 2019-01-15:
Same problem on RasPi 1 (hdmi), latest version.

Idowza, 2019-09-08:
Same issue here.  RasPi 3 B+

irovder, 2019-09-08:
Reinstalled Moode (latest for RasPi) and the issue is gone.

SimonCoulombe, 2020-02-22:
Just installed a fresh version of stretch on mi Pi2.  I added my spotify premium credentials.  

It works, but I also have to restart the service (sudo systemctl raspotify restart) every time I want to listen to something.

AdriDevelopsThings, 2021-02-16:
I have the same problem. I'm using the version `0.16.0~librespot.v0.1.3-37-g0adb851` and I've installed raspotify with apt.
Why is the issue closed even though exists? 
@dtcooper  Could you open the issue please?

SimonCoulombe, 2021-02-17:
I ended up setting a cronjob to reboot my pi every night at 4am and it has been rock solid.

AdriDevelopsThings, 2021-02-17:
I identified something:
The problem also happens while you are hearing music with raspotify. The music continues on the raspotify but the device will be removed in the device list in Spotify and the music in Spotify stops while raspotify continue playing the music. And I think you don’t have to restart your pi, you can simply restart the raspotify service, but it’s not a solution.

AdriDevelopsThings, 2021-02-25:
Ok, I think I found the problem: If your internet went away, you couldn't hear music, logically. But when the internet comes back, raspotify doesn't want to let you play music.  You have to restart raspotify after you reconnect to your network.

So the raspotify developers have to implement a restarting of the service when the internet has gone away.

download13, 2022-07-20:
```
[2022-07-19T06:27:16Z INFO  librespot_playback::player] Loading <fotka> with Spotify URI <spotify:track:6Xtcf7QFYmF4ye41fkpOh9>
[2022-07-19T06:27:16Z INFO  librespot_playback::player] <fotka> (121771 ms) loaded
[2022-07-19T06:27:46Z INFO  librespot_playback::player] This track may be subject to 1.29 dB of dynamic limiting at it's peak.
[2022-07-19T06:29:17Z INFO  librespot_playback::player] Loading <Snooze> with Spotify URI <spotify:track:2J3SlUak8rUmbqQKLhVsrx>
[2022-07-19T06:29:18Z INFO  librespot_playback::player] <Snooze> (115263 ms) loaded
[2022-07-19T06:31:13Z INFO  librespot_playback::player] Loading <Sleepless Night> with Spotify URI <spotify:track:19E1JLs1HA9og77YVVLenA>
[2022-07-19T06:31:13Z INFO  librespot_playback::player] <Sleepless Night> (109714 ms) loaded
[2022-07-19T10:59:33Z WARN  librespot_core::session] Connection to server closed.
[2022-07-19T10:59:33Z ERROR librespot_connect::spirc] subscription terminated
[2022-07-19T10:59:33Z WARN  librespot] Spirc shut down unexpectedly
[2022-07-19T10:59:33Z INFO  librespot_core::session] Connecting to AP "ap-guc3.spotify.com:4070"
[2022-07-19T10:59:34Z INFO  librespot_core::session] Authenticated as "[REMOVED]" !
[2022-07-19T10:59:34Z INFO  librespot_playback::mixer::softmixer] Mixing with softvol and volume control: Log(60.0)
[2022-07-19T10:59:34Z INFO  librespot_playback::convert] Converting with ditherer: tpdf
[2022-07-19T10:59:34Z INFO  librespot_playback::audio_backend::alsa] Using AlsaSink with format: S16
[2022-07-19T10:59:34Z INFO  librespot_core::session] Country: "US"
```

Here's a log of one instance if it helps. Judging by the `[2022-07-19T10:59:33Z WARN  librespot] Spirc shut down unexpectedly` line, I suspect @AdriDevelopsThings is right.

For now I'm just going to try to setup some kind of scheduled restart for the container every 4am or so.