dtcooper/raspotify #108: Raspotify doesn't show up as device after wifi reconnect

SirJohnDoe, 2018-04-29:
When wifi connection drops, naturally Raspotify dissappears as one of the Spotify devices, however, when wifi reconnects it still won't show up and it'll require a manual reboot to get it to work again.

Please fix this.

dtcooper, 2018-04-30:
The latest version of librespot may address this as far as I can tell. Try updating your package to 0.11.1 and let me know if it doesn't work.

closed 2018-04-30 by dtcooper

SirJohnDoe, 2018-04-30:
Unfortunately it doesn't seem to work. I've updated Raspotify, when it automatically started I could see it in Spotify, but when I manually disabled wifi and then enabled it again, it didn't attempt to reconnect. This is also clear from the log, which just shows the initial start-up (no mention of a reconnect).

> sudo service raspotify status
* raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-04-30 09:44:02 CEST; 3min 52s ago
  Process: 1303 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1300 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1305 (librespot)
   CGroup: /system.slice/raspotify.service
           `-1305 /usr/bin/librespot --name La Universidad --backend alsa --bitrate 320 --disable-audio-cache --initial-volume 100 --username --password

Apr 30 09:44:02 osmc-Willoughby systemd[1]: Starting Raspotify...
Apr 30 09:44:02 osmc-Willoughby systemd[1]: Started Raspotify.
Apr 30 09:44:02 osmc-Willoughby librespot[1305]: INFO:librespot: librespot (raspotify) 59cff3d (2018-04-25). Built on 2018-04-30. Build ID: 7BHFeq2k
Apr 30 09:44:02 osmc-Willoughby librespot[1305]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-v5wd.ap.spotify.com:4070"
Apr 30 09:44:03 osmc-Willoughby librespot[1305]: INFO:librespot_core::session: Authenticated as "something" !
Apr 30 09:44:03 osmc-Willoughby librespot[1305]: INFO:librespot_core::session: Country: "BE"
Apr 30 09:44:03 osmc-Willoughby librespot[1305]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink