dtcooper/raspotify #112: Spotify stops playing after a few songs

BugRogers42, 2018-05-17:
I'm using the android app to play music via a pi and you raspotify.
The Pi will be displayed as divice in the app and i can start the music. But after a few songs (can be 1, 2 or 10) it stops playing and the list of devices is sometimes empty, sometimes Raspotify is displayed as device but i can't connect. After restarting Raspotify it works again immediatly.
The Pi is cable-connected and I entered username/password in /etc/default/raspotify, but that doesn't work.

I know there are some similar issues but it is a very important one. Are you working on it?

regards

Heil-Pepito, 2018-05-17:
Hi, me too i have the same problem.
I wanted to open a new thread but I will answer here.

I tried to reinstall Raspbian and installing just Raspotify. It's doesn't work anymore.
Can connect (via Spotify Connect) at all and if i can connect, no music is played and i'm immedialty disconnected. Nothing is wrong with logins and service.
Few weeks ago it's work fine but since the last week no.

andy646362, 2018-05-17:
Same problem here. This is my error:

`* raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-05-17 19:36:52 CEST; 2s ago
  Process: 1848 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1845 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1852 (librespot)
   CGroup: /system.slice/raspotify.service
           `-1852 /usr/bin/librespot --name raspotify (osmc) --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --linear-

May 17 19:36:52 osmc systemd[1]: Starting Raspotify...
May 17 19:36:52 osmc systemd[1]: Started Raspotify.
May 17 19:36:52 osmc librespot[1852]: INFO:librespot: librespot (raspotify) 59cff3d (2018-04-25). Built on 2018-04-30. Build ID: 7BHFeq2k
May 17 19:36:52 osmc librespot[1852]: WARN:mdns: Failed to register IPv6 receiver: Os { code: 19, kind: Other, message: "No such device" }
May 17 19:36:53 osmc librespot[1852]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-6k5g.ap.spotify.com:4070"
May 17 19:36:53 osmc librespot[1852]: thread 'main' panicked at 'Authentication failed with reason: BadCredentials', core/src/connection/mod.rs:93:21
May 17 19:36:53 osmc librespot[1852]: stack backtrace:
May 17 19:36:54 osmc librespot[1852]:    0:   0x9ec333 - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h4ef6490c84cef3d0
`

I'm thinking this is a problem with Spotify itself.

gfro84, 2018-05-17:
I have an issue with similar effect. 

`● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-05-17 21:40:03 BST; 2min 44s ago
  Process: 574 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 572 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 578 (librespot)
   CGroup: /system.slice/raspotify.service
           └─578 /usr/bin/librespot --name RaPi_spotify --backend alsa --bitrate 320 --disable-audio-cache --initial-volume 65

May 17 21:40:03 osmc-Waldo librespot[578]: INFO:librespot: librespot (raspotify) 59cff3d (2018-04-25). Built on 2018-04-30. Build ID: 7BHFeq2k
May 17 21:40:03 osmc-Waldo librespot[578]: WARN:mdns: Failed to register IPv6 receiver: Os { code: 19, kind: Other, message: "No such device" }
May 17 21:41:55 osmc-Waldo librespot[578]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-ng9h.ap.spotify.com:4070"
May 17 21:41:56 osmc-Waldo librespot[578]: INFO:librespot_core::session: Authenticated as "my_user_name" !
May 17 21:41:56 osmc-Waldo librespot[578]: INFO:librespot_core::session: Country: "GB"
May 17 21:41:56 osmc-Waldo librespot[578]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
May 17 21:41:56 osmc-Waldo librespot[578]: INFO:librespot_playback::player: Loading track "Malibu" with Spotify URI "spotify:track:1UZOjK1BwmwWU14Erba9CZ"
May 17 21:41:57 osmc-Waldo librespot[578]: INFO:librespot_playback::player: Track "Malibu" loaded
May 17 21:42:46 osmc-Waldo librespot[578]: INFO:librespot_playback::player: Loading track "Pick Up" with Spotify URI "spotify:track:3FKMngtZtIZCNKYexhGrVD"
May 17 21:42:47 osmc-Waldo librespot[578]: INFO:librespot_playback::player: Track "Pick Up" loaded
osmc@osmc-Waldo:~$ service raspotify status
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2018-05-17 21:43:47 BST; 3s ago
  Process: 578 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS (code=exited, status=101)
  Process: 574 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 572 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 578 (code=exited, status=101)

May 17 21:43:47 osmc-Waldo systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
May 17 21:43:47 osmc-Waldo systemd[1]: raspotify.service: Unit entered failed state.
May 17 21:43:47 osmc-Waldo systemd[1]: raspotify.service: Failed with result 'exit-code'.
`

Heil-Pepito, 2018-05-18:
I tried to use Raspotify and Spotifyd but nothing work. Both use Librespot so i launched this alone and he give me some errors. (Segmentation Fault)
I wanted to install his fork (https://github.com/librespot-org/librespot) but I do not know how to recover the repository and compile. If anyone knows how to do and test, it will be nice to keep up to date.

dtcooper, 2018-05-18:
@Heil-Pepito: I'm already using the librespot-org fork, but if you'd like to try building against a fork of librespot, try the [instructions I laid out here](https://github.com/dtcooper/raspotify/issues/55#issuecomment-354187018). Thanks.

Heil-Pepito, 2018-05-18:
Thanks dtcooper, i've fixed my Librespot problem but the initial problem stay here.

BugRogers42, 2018-05-19:
2 days ago there was a new update of the andoid app. Now raspotify doesn't work anymore for me. In the app i can see the device for one second then it disappears again. When i restart raspotify it takes some time then the same effect happens again.

JeremieRapin, 2018-05-19:
I have the same behavior and an ugly fix. Add a crontab entry which survey the raspotify's status.

Create a file [watchdog.sh](https://gist.github.com/JeremieRapin/bc35a2632c072a082d48f5503270df16) in /home/pi.
Then add an entry in /etc/crontab, as described in the script's comment .

Raspotify still crashing, but in the next minute, it is relaunch and music can be play again.

My 2 cents, ....

andy646362, 2018-05-19:
@dtcooper My issue was resolved by the latest update, thanks!

ruitome, 2018-05-19:
I'm having the same issue as @BugRogers42. I can select the Pi in the spotify client, and if I want to change the volume level it crashes immediately. If I don't change anything it crashes a few seconds after playing. This is happening since 2 days ago.

scarapunzi, 2018-05-20:
Same issue of @BugRogers42 here!

gfro84, 2018-05-21:
See #114 - update raspotify and you will be fixed

ruitome, 2018-05-21:
Thanks @gfro84! This worked for me :)

Heil-Pepito, 2018-05-21:
@dtcooper It's working now without problems but i can't confirm if it's solved by the update. In all cases, thanks for your help.

nomoreme, 2018-05-28:
Hi, I have the same problem described in the first post by @BugRogers42. I can see the Pi in the device list in app and sometimes I can start the music and after a few songs it stops, sometimes spotify won't connect to the Pi and the list became empty.
After a Pi restart (and some relaunch of the spotify app on my phone), everything works.

Everything is updated with `sudo apt-get update && sudo apt-get upgrade`.

Here's the log from `sudo systemctl status raspotify`:

```
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-05-27 16:11:16 UTC; 1 day 5h ago
  Process: 481 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 470 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 483 (librespot)
   CGroup: /system.slice/raspotify.service
           └─483 /usr/bin/librespot --name raspotify (raspberrypi) --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=

May 27 17:00:52 raspberrypi librespot[483]: INFO:librespot_playback::player: Track "Miracle" loaded
May 27 17:04:00 raspberrypi librespot[483]: INFO:librespot_playback::player: Loading track "Cloud Nine" with Spotify URI "spotify:track:34nw5BCrCKN6nVrL4Zaong"
May 27 17:04:01 raspberrypi librespot[483]: INFO:librespot_playback::player: Track "Cloud Nine" loaded
May 27 17:07:53 raspberrypi librespot[483]: INFO:librespot_playback::player: Loading track "One Kiss (with Dua Lipa)" with Spotify URI "spotify:track:7ef4DlsgrMEH11cDZd32M6"
May 27 17:07:55 raspberrypi librespot[483]: INFO:librespot_playback::player: Track "One Kiss (with Dua Lipa)" loaded
May 27 17:11:29 raspberrypi librespot[483]: INFO:librespot_playback::player: Loading track "Flames" with Spotify URI "spotify:track:33IOhptvC2Qoy2UhjiHXLV"
May 27 17:11:30 raspberrypi librespot[483]: INFO:librespot_playback::player: Track "Flames" loaded
May 27 17:14:45 raspberrypi librespot[483]: INFO:librespot_playback::player: Loading track "You Owe Me" with Spotify URI "spotify:track:3ZBksMOUTbOTIgIwV0RaLg"
May 27 17:14:46 raspberrypi librespot[483]: INFO:librespot_playback::player: Track "You Owe Me" loaded
May 27 17:17:57 raspberrypi librespot[483]: ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
```

Let me know if you need more debugging info.
And thanks for your help (and the amazing client... when it works is <3 )

rodet, 2018-05-31:
I had a similar problem for a few weeks and was wondering what was happening. Yesterday my Raspberry went out of disk space. After making space again on the Raspberry (`/var/log`alone was taking 1.5GB), Raspotify is working now perfectly. I guess Raspotify has some minimum space requirement for caching files on the disk? Anyway, it's quick looking for how much space you still have on your device and trying to clean up a bit.
@dtcooper maybe something to add to the troubleshooting section?

dtcooper, 2018-05-31:
I'll take a look at what happens in `/var/log`. Could you folks check to see if your disk/SD card is full?

rodet, 2018-05-31:
Sorry for the misunderstanding... The size of the logs must have been from Raspbian overall, not from Raspotify (I assume)

nomoreme, 2018-06-02:
I don't think I have any space isuue:

Filesystem      Size  Used Avail Use% Mounted on                                       
/dev/root        30G  3.0G   26G  11% /                                                
devtmpfs        460M     0  460M   0% /dev                                             
tmpfs           464M  8.0K  464M   1% /dev/shm                                         
tmpfs           464M  6.2M  458M   2% /run                                             
tmpfs           5.0M  4.0K  5.0M   1% /run/lock                                        
tmpfs           464M     0  464M   0% /sys/fs/cgroup                                   
/dev/mmcblk0p1   41M   22M   20M  53% /boot                                            
/dev/sda1       299G  149G  150G  50% /mnt/passport                                    
tmpfs            93M     0   93M   0% /run/user/1000

BugRogers42, 2018-06-12:
ok, so the last update fixed the problem for me. But a few days ago spotify relesed another update and now raspotify again disapears immediately in spotify. Also when i restart the service.

Rainee4563, 2018-06-12:
I was able to fix it by installing it along side spotify web connect

teddycool, 2018-07-27:
Hi,
I have this problem as well on a fresh install of 2018-06-27-raspbian-stretch-lite on a raspberry pi 3.
I agree that this makes the raspotify allmost useless. It plays a couple of tracks and then disconnects. 
A restart of the service solves the problem though.
I have not filled the sdcard, its alot of free space. I have played with the settings but for me it hasn't changed the behaivour. 
In the future I will also use the spotify web api from python (spotipy) to show artist and track names on a display and maybe that solves the issue as it did for Rainee4563 .
In case I can contribute with logfiles or tests to help solve the prorblem, please let me know.

teddycool, 2018-08-03:
In the log it hints about a spotify problem (reset by peer) but this is not the case since I have several spotify connect units in the same network that works perfecly for hours and hours (volumio with spotify connect on the same type of pi is one example). However I would like to use Raspotify instead since it's a service running in the background without any 'fuzz' or  extras...   But maybe it dosn't like Motorhead... ;-)
Jul 28 07:51:22 spotipi librespot[498]: INFO:librespot_core::session: Country: "SE"
Jul 29 17:21:07 spotipi librespot[498]: INFO:librespot_playback::player: Loading track "Cat Scratch Fever" with Spotify URI "spotify:track:1uH06nM0HEsbCqvdaqzwa5"
Jul 29 17:21:09 spotipi librespot[498]: INFO:librespot_playback::player: Track "Cat Scratch Fever" loaded
Jul 29 17:22:23 spotipi librespot[498]: INFO:librespot_playback::player: Loading track "Killed by Death" with Spotify URI "spotify:track:4W2CRPuOl8wN3aR7QmOmQE"
Jul 29 17:22:25 spotipi librespot[498]: INFO:librespot_playback::player: Track "Killed by Death" loaded
Jul 29 17:23:15 spotipi librespot[498]: INFO:librespot_playback::player: Loading track "Electricity" with Spotify URI "spotify:track:1KYfhcM7K7Ahs6opoG4cd0"
Jul 29 17:23:17 spotipi librespot[498]: INFO:librespot_playback::player: Track "Electricity" loaded
Jul 29 17:25:33 spotipi librespot[498]: INFO:librespot_playback::player: Loading track "Evil Eye" with Spotify URI "spotify:track:1jsZ5wEBDUioHIWvOmHP4g"
Jul 29 17:25:35 spotipi librespot[498]: INFO:librespot_playback::player: Track "Evil Eye" loaded
Jul 29 17:27:55 spotipi librespot[498]: ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }

abrakadabra2k, 2018-09-12:
Any update on this issue? I am seeing the same behavior on the most recent moobe built. (and also on volumio, by the way. Connecting to a chrome cast works perfectly, on the other hand)

romainp, 2018-09-25:
Same error. Stops playing after a couple songs. 

`
Sep 25 20:26:33 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Sep 25 20:27:23 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Sep 25 20:28:13 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Sep 25 20:29:03 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Sep 25 20:29:53 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Sep 25 20:30:43 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Sep 25 20:31:32 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Sep 25 20:32:22 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Sep 25 20:33:12 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Sep 25 20:34:01 raspberrypi librespot[1025]: WARN:mdns::fsm: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }`

romainp, 2018-10-01:
removing pulseaudio solved the issue for me! Working great now, love it!

warro, 2019-02-20:
Same problem here.

romainp, 2019-02-21:
> Same problem here.

Have you tried to remove pulseaudio?

lamoe02, 2019-02-28:
noob here so please be gentle :)
what I think solved the 104 error for me was to delete all spotify related files on my android phone(under settings, apps, spotify, delete app data). I don't know if this makes sense and is in any way connected... but no updates etc helped. after deleting the files on my phone I had to login again to the spotify app on my phone and ever since then no more raspotify problems...  

I had the disconnect issue (
code: 104, kind: ConnectionReset, message: "Connection reset by peer) "for months and nothing helped, no update, restart, new install etc. 
I'm running raspotify on a raspberry pi first generation with raspbian stretch.
It might be magic or luck, I don't know. 
But I thought I share this in case it helps others as well?

Brucki, 2019-03-04:
Have you tried to disable any battery saving modes?
Had the same problem with my HUAWEI phone, it's working now since i disabled optimisations for the Spotify app.

Settings -> Battery -> Launch -> Spotify ... Manage manually & allow all

ljubadr, 2019-07-12:
I had the same issue for months and I finally managed to make it work. Issue was not with the raspotify, but with the PulseAudio and HDMI. This solution will work only if you use HDMI cable for the sound.

On the PulseAudio/Troubleshooting page I found this:
[No HDMI sound output after some time with the monitor turned off](https://wiki.archlinux.org/index.php/PulseAudio/Troubleshooting#No_HDMI_sound_using_a_headless_server
)
> By default, this signal is turned off after 600 seconds, thus the audio sink gets lost as well.

In your pi terminal or ssh, run command to see the current time

```
cat /sys/module/kernel/parameters/consoleblank
```

For me it was `600`, and I used this solution to change it

[Console goes blank after a while](https://www.raspberrypi.org/forums/viewtopic.php?t=122140#p823537
)

Edit file `sudo vi /bott/cmdline.txt` and add `consoleblank=0` to the end of the line.

Double check if change was applied
```
cat /sys/module/kernel/parameters/consoleblank
```
I rebooted pi after the change for other reasons, so I'm not sure if needed.

If the first solution doesn't work for you, maybe check [Disable screen blanking on text console](https://unix.stackexchange.com/questions/8056/disable-screen-blanking-on-text-console/32043)

Good luck

elagil, 2020-01-09:
> I have the same behavior and an ugly fix. Add a crontab entry which survey the raspotify's status.
>
> Create a file [watchdog.sh](https://gist.github.com/JeremieRapin/bc35a2632c072a082d48f5503270df16) [...]

A small change to the script:
https://gist.github.com/elagil/dd32745a593fb6161579ae0926045e30

This takes into account the last minute of logs. Sometimes the normal status message does not include the "WARN" or "ERROR" statements, if the stack trace is too long.

jkwcht, 2020-04-03:
I am running raspotify on a Raspberry Pi 4 with Raspbian Buster and I also had the Error `librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }`
I could resolve it by not using wifi but ethernet. Wifi goes to standby after some time and raspotify can not handle that. 
The other issue regarding the audio stop working 5 minutes after reboot was due to using audio via HDMI and that raspbian blanks the screen (and audio with it) after 600 seconds by default. So install a GUI screensaver and deactivate it afterwards.

igno2k, 2020-05-31:
librespot cannot handle the wifi standby.
check it with:
`iw wlan0 get power_save`

You can disable power saving one time with:
`iw wlan0 set power_save off`

Or disable it permanently with a udev rule. My example for Arch Linux:
file: ´/etc/udev/rules.d/81-wifi-powersave.rules´:
´ACTION=="add", SUBSYSTEM=="net", KERNEL=="wl*", RUN+="/usr/bin/iw dev $name set power_save off"´

rainer-schreiber, 2020-12-05:
> I am running raspotify on a Raspberry Pi 4 with Raspbian Buster and I also had the Error `librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }`
> I [...]

Man, did you save my day @jkwcht. I was pulling my hair off for one whole day, and then I found out that after I touched my mouse, the sound came back :) So no need for installing a newer version of librespot, but just Disable Screen Blanking under Raspberry Pi Configuration -> Display tab.

JasonLG1979, 2021-11-24:
I'm pretty sure this is fixed upstream. I anyone is still having this same issue on the most current Raspotify package feel free to reopen this.

closed 2021-11-24 by JasonLG1979

fennovde, 2021-12-23:
I've still got the same error, however my raspberry is connected via LAN. I've disabled the power saving anyway, that didn't help.
I'm struggling with this problem for quite some time now however I haven't found a solution.
[image]
.

JasonLG1979, 2021-12-23:
> I've still got the same error, however my raspberry is connected via LAN. I've disabled the power saving anyway, that didn't help. I'm struggling with this problem for quite some time now however I haven't found a solution. [image] .

Raspotify is a packaged version of librespot. Any issues unrelated to packaging are upstream issues.

JasonLG1979, 2021-12-23:
Anecdotally I had problems with `librespot` dropping so I switched my router I'm not exactly sure what the issue is but the mDNS implementation in `librespot`, [libmdns](https://github.com/librespot-org/libmdns) could use some smoothing of rough edges.