dtcooper/raspotify #113: Raspotify stopped working

analogic, 2018-05-17:
From some strange reason raspotify stopped working entirely. It shows at list but when I hit play it crash with this message:

```
May 17 16:11:24 raspberrypi systemd[1]: Started Raspotify.
May 17 16:11:24 raspberrypi librespot[1326]: INFO:librespot: librespot (raspotify) 59cff3d (2018-04-25). Built on 2018-04-30. Build ID: 7BHFeq2k
May 17 16:11:24 raspberrypi librespot[1326]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-kl7n.ap.spotify.com:4070"
May 17 16:11:25 raspberrypi librespot[1326]: INFO:librespot_core::session: Authenticated as "...." !
May 17 16:11:25 raspberrypi librespot[1326]: INFO:librespot_core::session: Country: "CZ"
May 17 16:11:25 raspberrypi librespot[1326]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
May 17 16:11:25 raspberrypi librespot[1326]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: WireError(InvalidEnumValue(14))', libcore
May 17 16:11:25 raspberrypi librespot[1326]: stack backtrace:
May 17 16:11:26 raspberrypi librespot[1326]:    0:   0x9c4333 - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h4ef6490c84cef3d0
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
May 17 16:11:26 raspberrypi librespot[1326]:    1:   0x9be08f - std::sys_common::backtrace::_print::hea150ee4f33bb5de
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/sys_common/backtrace.rs:71
May 17 16:11:26 raspberrypi librespot[1326]:    2:   0x9c2da3 - std::panicking::default_hook::{{closure}}::h6c4396ff5bd61d89
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/sys_common/backtrace.rs:59
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/panicking.rs:380
May 17 16:11:26 raspberrypi librespot[1326]:    3:   0x9c2987 - std::panicking::default_hook::h0f6cb16cf06a0198
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/panicking.rs:396
May 17 16:11:26 raspberrypi librespot[1326]:    4:   0x9c3247 - std::panicking::rust_panic_with_hook::h2207195920ffa55f
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/panicking.rs:576
May 17 16:11:26 raspberrypi librespot[1326]:    5:   0x9c30ab - std::panicking::begin_panic::h8d43a09cf6d6e5f1
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/panicking.rs:537
May 17 16:11:26 raspberrypi librespot[1326]:    6:   0x9c302f - std::panicking::begin_panic_fmt::hd9afc94159ee9735
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/panicking.rs:521
May 17 16:11:26 raspberrypi librespot[1326]:    7:   0x9c2fc7 - rust_begin_unwind
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/panicking.rs:497
May 17 16:11:26 raspberrypi librespot[1326]:    8:   0xa10e57 - core::panicking::panic_fmt::h0bd854df201d1baf
May 17 16:11:26 raspberrypi librespot[1326]:                        at libcore/panicking.rs:71
May 17 16:11:26 raspberrypi librespot[1326]:    9:   0x53ff4b - core::result::unwrap_failed::h98683af3bb449e2d
May 17 16:11:26 raspberrypi librespot[1326]:   10:   0x51f20f - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &'a mut F>::call_once::h3
May 17 16:11:26 raspberrypi librespot[1326]:   11:   0x554ec7 - <futures::stream::map::Map<S, F> as futures::stream::Stream>::poll::h0581f95c62f44682
May 17 16:11:26 raspberrypi librespot[1326]:   12:   0x522137 - <librespot_connect::spirc::SpircTask as futures::future::Future>::poll::he9eb166b9c820dcd
May 17 16:11:26 raspberrypi librespot[1326]:   13:   0x50648b - <librespot::Main as futures::future::Future>::poll::ha710f437106cfed1
May 17 16:11:26 raspberrypi librespot[1326]:   14:   0x4ffc07 - futures::task_impl::std::set::h290f5a43ee28e3de
May 17 16:11:26 raspberrypi librespot[1326]:   15:   0x4f6c1b - <futures::future::lazy::Lazy<F, R> as futures::future::Future>::poll::hf2b09d2752d9a9f9
May 17 16:11:26 raspberrypi librespot[1326]:   16:   0x4fff73 - futures::task_impl::std::set::hb747992eb6246e3b
May 17 16:11:26 raspberrypi librespot[1326]:   17:   0x4eabd3 - <std::thread::local::LocalKey<T>>::with::hf98e87107911adf9
May 17 16:11:26 raspberrypi librespot[1326]:   18:   0x4ed2f7 - <tokio::executor::current_thread::Entered<'a, P>>::block_on::hd350dc4f0d2f17e2
May 17 16:11:26 raspberrypi librespot[1326]:   19:   0x4ea613 - <std::thread::local::LocalKey<T>>::with::h7a0beeac83672a4f
May 17 16:11:26 raspberrypi librespot[1326]:   20:   0x4ea87f - <std::thread::local::LocalKey<T>>::with::hc8cef14402e834e4
May 17 16:11:26 raspberrypi librespot[1326]:   21:   0x4e864f - <scoped_tls::ScopedKey<T>>::set::h90031fa39bc1a6fb
May 17 16:11:26 raspberrypi librespot[1326]:   22:   0x4e4153 - tokio_core::reactor::Core::run::h09af0e02db6cafcd
May 17 16:11:26 raspberrypi librespot[1326]:   23:   0x50823b - librespot::main::h2aac80d528d4c194
May 17 16:11:26 raspberrypi librespot[1326]:   24:   0x4f580b - std::rt::lang_start::{{closure}}::hecf37fca32271fe5
May 17 16:11:26 raspberrypi librespot[1326]:   25:   0x9c2eff - std::panicking::try::do_call::h5f03c49f73b7300e
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/rt.rs:59
May 17 16:11:26 raspberrypi librespot[1326]:                        at libstd/panicking.rs:479
May 17 16:11:26 raspberrypi librespot[1326]:   26:   0x9d2673 - __rust_maybe_catch_panic
May 17 16:11:27 raspberrypi librespot[1326]:                        at libpanic_unwind/lib.rs:102
May 17 16:11:27 raspberrypi librespot[1326]:   27:   0x9c366f - std::rt::lang_start_internal::h907d77b40aa767dd
May 17 16:11:27 raspberrypi librespot[1326]:                        at libstd/panicking.rs:458
May 17 16:11:27 raspberrypi librespot[1326]:                        at libstd/panic.rs:358
May 17 16:11:27 raspberrypi librespot[1326]:                        at libstd/rt.rs:58
May 17 16:11:27 raspberrypi librespot[1326]:   28:   0x508e23 - main
May 17 16:11:27 raspberrypi librespot[1326]:   29: 0xb6cc1677 - __libc_start_main
```

analogic, 2018-05-17:
Ha, sorry it seems librespot bug

closed 2018-05-17 by analogic

dtcooper, 2018-05-17:
No problem. Thanks for filing. This has been fixed in Raspotify. See #114 :)