dtcooper/raspotify #114: Raspotify Service Exited after connecting to device

e-3-s, 2018-05-17:
Hi there,

The raspotify service is giving me an error.
Sometimes its works sometimes it doesn't

`● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2018-05-17 23:00:06 CEST; 2s ago
  Process: 1022 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS (code=exited, status=101)
  Process: 1020 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1017 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1022 (code=exited, status=101)

May 17 23:00:06 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
May 17 23:00:06 raspberrypi systemd[1]: raspotify.service: Unit entered failed state.
May 17 23:00:06 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
`

mim999, 2018-05-17:
Had no problems with raspotify until today, uninstalled en reinstalled it. but no success. have the same error now.

dtcooper, 2018-05-17:
This seems to be related to [Librespoty's issue #233](https://github.com/librespot-org/librespot/issues/223). Tracking that and attempting a fix now. Thanks

dtcooper, 2018-05-17:
Going to use @dawidd6's branch to see if that fix works. Building now!

gfro84, 2018-05-17:
@dtcooper let us know how you get on with the @dawidd6 fork build!

dtcooper, 2018-05-17:
The fork works! The new version of Raspotify is pushed to the apt repo. This should fix the problem, but let me know if you have issues!

```
sudo apt-get update
sudo apt-get upgrade
```

closed 2018-05-17 by dtcooper

gfro84, 2018-05-17:
Brilliant - works a treat. For any folk on OSMC, need to re-run the install.sh as 
sudo apt-get upgrade 
Results in a message telling you you are doing it wrong. Install.sh pulls the new file from the repository.

@dtcooper thanks for the quick fix

mim999, 2018-05-17:
@dtcooper thanks you for the quick fix!

dtcooper, 2018-05-17:
No problem, folks! 🙃

e-3-s, 2018-05-18:
Wauw just coming home. And its already fixed.
You sure are great!