dtcooper/raspotify #115: Crash at each startup

JeremieRapin, 2018-05-18:
Each time, librespot is started, it crashes with the following traces.

```
/usr/bin/librespot --name name --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --username user --password password
INFO:librespot: librespot (raspotify) 59cff3d (2018-04-25). Built on 2018-04-30. Build ID: 7BHFeq2k
INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-xzdx.ap.spotify.com:4070"
INFO:librespot_core::session: Authenticated as "user" !
INFO:librespot_playback::audio_backend::alsa: Using alsa sink
INFO:librespot_core::session: Country: "FR"
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: WireError(InvalidEnumValue(14))', libcore/result.rs:945:5
stack backtrace:
   0:   0x9a7333 - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h4ef6490c84cef3d0
                       at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
   1:   0x9a108f - std::sys_common::backtrace::_print::hea150ee4f33bb5de
                       at libstd/sys_common/backtrace.rs:71
   2:   0x9a5da3 - std::panicking::default_hook::{{closure}}::h6c4396ff5bd61d89
                       at libstd/sys_common/backtrace.rs:59
                       at libstd/panicking.rs:380
   3:   0x9a5987 - std::panicking::default_hook::h0f6cb16cf06a0198
                       at libstd/panicking.rs:396
   4:   0x9a6247 - std::panicking::rust_panic_with_hook::h2207195920ffa55f
                       at libstd/panicking.rs:576
   5:   0x9a60ab - std::panicking::begin_panic::h8d43a09cf6d6e5f1
                       at libstd/panicking.rs:537
   6:   0x9a602f - std::panicking::begin_panic_fmt::hd9afc94159ee9735
                       at libstd/panicking.rs:521
   7:   0x9a5fc7 - rust_begin_unwind
                       at libstd/panicking.rs:497
   8:   0x9f3e57 - core::panicking::panic_fmt::h0bd854df201d1baf
                       at libcore/panicking.rs:71
   9:   0x522f4b - core::result::unwrap_failed::h98683af3bb449e2d
  10:   0x50220f - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &'a mut F>::call_once::h3bf172110ebe5c0b
  11:   0x537ec7 - <futures::stream::map::Map<S, F> as futures::stream::Stream>::poll::h0581f95c62f44682
  12:   0x505137 - <librespot_connect::spirc::SpircTask as futures::future::Future>::poll::he9eb166b9c820dcd
  13:   0x4e948b - <librespot::Main as futures::future::Future>::poll::ha710f437106cfed1
  14:   0x4e2c07 - futures::task_impl::std::set::h290f5a43ee28e3de
  15:   0x4d9c1b - <futures::future::lazy::Lazy<F, R> as futures::future::Future>::poll::hf2b09d2752d9a9f9
  16:   0x4e2f73 - futures::task_impl::std::set::hb747992eb6246e3b
  17:   0x4cdbd3 - <std::thread::local::LocalKey<T>>::with::hf98e87107911adf9
  18:   0x4d02f7 - <tokio::executor::current_thread::Entered<'a, P>>::block_on::hd350dc4f0d2f17e2
  19:   0x4cd613 - <std::thread::local::LocalKey<T>>::with::h7a0beeac83672a4f
  20:   0x4cd87f - <std::thread::local::LocalKey<T>>::with::hc8cef14402e834e4
  21:   0x4cb64f - <scoped_tls::ScopedKey<T>>::set::h90031fa39bc1a6fb
  22:   0x4c7153 - tokio_core::reactor::Core::run::h09af0e02db6cafcd
  23:   0x4eb23b - librespot::main::h2aac80d528d4c194
  24:   0x4d880b - std::rt::lang_start::{{closure}}::hecf37fca32271fe5
  25:   0x9a5eff - std::panicking::try::do_call::h5f03c49f73b7300e
                       at libstd/rt.rs:59
                       at libstd/panicking.rs:479
  26:   0x9b5673 - __rust_maybe_catch_panic
                       at libpanic_unwind/lib.rs:102
  27:   0x9a666f - std::rt::lang_start_internal::h907d77b40aa767dd
                       at libstd/panicking.rs:458
                       at libstd/panic.rs:358
                       at libstd/rt.rs:58
  28:   0x4ebe23 - main
  29: 0x76d69677 - __libc_start_main
```

Adr921, 2018-05-18:
Same here :( Thanks for opening the issue!

JeremieRapin, 2018-05-18:
Yes, fixed in #114 .
Ticket could be closed.

closed 2018-05-18 by JeremieRapin

dtcooper, 2018-05-18:
Sorry folks. Spotify's fault, not ours. It was addressed upstream in [librespot](https://github.com/librespot-org/librespot). Update the package and it should be fixed.