dtcooper/raspotify #116: raspotify device available on two account with config set to one

mim999, 2018-05-20:
I have a script running that allows raspotify on my raspberry pi to switch from account by changing the CONFIG settings. This has worked until today. Now I can still see and select the device even though it is set to someone elses device.

primohacker, 2018-06-05:
You can actually run raspotify from any account in the same subnet. Credentialed setup allows the raspotify device to be used from other subnets. Knowing this, credentials may be needed when using a VPN, depending on the tunnel and route table.

I had also created a script to swap creds, until I noticed this behavior.

terrar, 2018-10-05:
Am i able to disable this feature so that only my account is able to use the raspotify device?

closed 2021-11-24 by github-actions[bot]