dtcooper/raspotify #119: Needs to be restarted erverytime

Ptafke, 2018-05-30:
I have to restart every time I want to use it. Bevor it is not found. 
Raspbian.
Version 0.11.2

Thank you.

tjvr, 2018-05-30:
What do you see in the logs? Try `sudo journalctl -ef` perhaps

Ptafke, 2018-06-01:
There is the Problem. The logs showing nothing. For example systemctl status shows `Warning: Journal has been rotated since unit was started. Log output is incomple...`
and journalctl is emtpy:
`~ $ journalctl -u raspotify.service
-- No entries --`

The global journal is floated with:
 ```
~$  sudo journalctl -ef
....
Jun 01 11:42:36 licht lircd[8958]: lircd-0.9.4c[8958]: Error: Cannot glob /sys/class/rc/rc0/input[0-9]*/event[0-9]*
....
```
But mybe this is my Problem. I found there only the restart message. After restarting all is fine.

Thank you for your input.

closed 2021-11-24 by github-actions[bot]