dtcooper/raspotify #121: Bad credentials

scarapunzi, 2018-05-31:
Hi! I have manually set my username and password in the /etc/default/raspotify as described, but I keep getting a panic on "Bad Credentials".
I'm sure credentials are correct, I think the problem exists because my username is surrounded by two * (like this: *username*). So I have set it correctly but I always get a bad login. Spotify gives me no way to change the username (the only way would be to create a new account). Any fix would be appreciated. Thank you

dtcooper, 2018-05-31:
Hey @scarapunzi, I think you'd be better served by filing this as a bug with [librespot](https://github.com/librespot-org/librespot), as I believe it would be a bug with that. In any event, if you post the `systemd` logs with your credentials redacted, but with the asterisks still in place I'll have a look. Thanks!

scarapunzi, 2018-05-31:
Hi @dtcooper first of all thank you for your answer and help!

Here my raspotify config file:
# /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
#DEVICE_NAME="raspotify"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="320"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username <USERNAME> --password <PASSWORD>"
OPTIONS="--username <*xxxxxx*> --password <xxxxxxx> --device hw:1,0"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
#VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"
VOLUME_ARGS="--enable-volume-normalisation --initial-volume=70"

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"

Here result of sudo systemctl status raspotify:

scarapunzi, 2018-06-01:
I have done other tests: the problem in my opinion is that you should implement in your config separate hints for USERNAME and PASSWORD. It's only an interpretation problem by librespot cause of special characters!

Now your config is set in this way:
OPTIONS="--username "user" --password "password" --device hw:1,0""

To solve in my opinion you should implement separate hints for username and password like this:

USERNAME="username"
PASSWORD="password"
OPTIONS="--device hw:1,0"

Thank you for your help!

scarapunzi, 2018-06-03:
Hey @dtcooper, sorry for insistence, but do you have any idea about my suggestion?
Discovery mode in my local network doesn't work very well, often I can't see raspotify client from my android phone (raspberry and phone are on the same network)!
To make it discoverable, every time I power on my raspberry I always have to disconnect and reconnect wifi several times on my phone, if I don't do it raspotify client doesn't show up.
I could solve setting up user and pass, but I have the problem of special characters.
I'm sure it's only a problem of interpretation by librespot and could be simple solved as I posted before.

Thank you again for your patience

closed 2021-11-24 by github-actions[bot]