dtcooper/raspotify #126: Can choose it in Spotify but it doesn't connect

milyglos, 2018-06-21:
Hi,

I am totally new in this. I have a RPi3 with Hifiberry. Used to use musicbox but it frequently didn't work and today installed Raspbian and Raspotify... Well... It even worked. Once. I decided to switch the RPi off to put it back into the case and now it doesn't work anymore. I have seen one post with the same issue but.... I don't know. I can see the "Malinka" (small raspberry in Polish) listed in Spotify Connect list but wen I click it it says "Connecting" and after couple of seconds it goes "Standby" again... 
I've installed the Raspotify with the auto option and edited the conf... Help? :)

All best!

Kamil

stefjan, 2018-06-24:
Hi Kamil

i had a similar problem: raspotify showed up in the Spotify connect list, but i could not connect (connecting -> standby). The problem was that i had port 4070 blocked in my firewall. Maybe you have a similar problem?

Have a look at the file /var/log/daemon.log . you should see messages like  'Started Raspotify.' 'INFO:librespot_core::session: Connecting to AP "XXX:4070"' (where XXX is some spotify.com address), 'INFO:librespot_core::session: Authenticated as "YYY"' (where YYY is your spotify username - in case you filled it in in the conf-file).

i did not see that last line at the start, but after i tried to connect via the spotify connect list there were messages like 'thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 110, kind: TimedOut, message: "Connection timed out" }', libcore/result.rs:945:5' (you will need to be patient for them to show up, it takes quite some time).

Opening up port 4070 solved it for me.

Hope it helps
stefan

tobiasmcnulty, 2018-06-24:
raspotify recently stopped working for me, too. Not sure if this is the issue you're seeing, but looking at `/var/log/daemon.log` there is a failure (not firewall related). The process fails, exits, and get restarted by systemd:

    Jun 24 21:35:35 pi0 librespot[628]: INFO:librespot_core::session: Connecting to AP "guc3-accesspoint-b-7p06.ap.spotify.com:4070"
    Jun 24 21:35:35 pi0 librespot[628]: INFO:librespot_core::session: Authenticated as "<snip>" !
    Jun 24 21:35:35 pi0 librespot[628]: INFO:librespot_core::session: Country: "US"
    Jun 24 21:35:35 pi0 librespot[628]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
    Jun 24 21:35:36 pi0 librespot[628]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: SpotifyIdError', libcore/result.rs:945:5
    Jun 24 21:35:36 pi0 librespot[628]: stack backtrace:
    Jun 24 21:35:36 pi0 librespot[628]:    0:   0x985d1b - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::hd7376efaf737a6cb
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
    Jun 24 21:35:36 pi0 librespot[628]:    1:   0x98cd63 - std::sys_common::backtrace::print::h0b8b4ef073b17e82
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/sys_common/backtrace.rs:71
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/sys_common/backtrace.rs:59
    Jun 24 21:35:36 pi0 librespot[628]:    2:   0x97c323 - std::panicking::default_hook::{{closure}}::h984bb18536e1dcf9
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/panicking.rs:207
    Jun 24 21:35:36 pi0 librespot[628]:    3:   0x97bf77 - std::panicking::default_hook::h637e82b93eb77513
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/panicking.rs:223
    Jun 24 21:35:36 pi0 librespot[628]:    4:   0x97c7cb - std::panicking::rust_panic_with_hook::h4c1dc13cc6e725f4
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/panicking.rs:402
    Jun 24 21:35:36 pi0 librespot[628]:    5:   0x97c5e7 - std::panicking::begin_panic_fmt::he64fbc36f455b25f
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/panicking.rs:349
    Jun 24 21:35:36 pi0 librespot[628]:    6:   0x97c53f - rust_begin_unwind
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/panicking.rs:325
    Jun 24 21:35:36 pi0 librespot[628]:    7:   0x9d43d7 - core::panicking::panic_fmt::h2bc4e257e1a74d55
    Jun 24 21:35:36 pi0 librespot[628]:                        at libcore/panicking.rs:72
    Jun 24 21:35:36 pi0 librespot[628]:    8:   0x4ecb5f - core::result::unwrap_failed::h05e9fd56219f727b
    Jun 24 21:35:36 pi0 librespot[628]:    9:   0x4e7ceb - librespot_connect::spirc::SpircTask::load_track::h142c92b868e5907d
    Jun 24 21:35:36 pi0 librespot[628]:   10:   0x4e66cb - <librespot_connect::spirc::SpircTask as futures::future::Future>::poll::h02d86f1bf0871f65
    Jun 24 21:35:36 pi0 librespot[628]:   11:   0x4c55d7 - <librespot::Main as futures::future::Future>::poll::h1bff5ea08bb1fc71
    Jun 24 21:35:36 pi0 librespot[628]:   12:   0x49e73b - <futures::task_impl::Spawn<T>>::poll_future_notify::h2e97edddde4b9ea4
    Jun 24 21:35:36 pi0 librespot[628]:   13:   0x49e53b - <futures::task_impl::Spawn<T>>::poll_future_notify::h1b9f2b276f42e8e3
    Jun 24 21:35:36 pi0 librespot[628]:   14:   0x4a871b - <std::thread::local::LocalKey<T>>::with::h1ead26e3b2d8ac2b
    Jun 24 21:35:36 pi0 librespot[628]:   15:   0x4b520f - <tokio::executor::current_thread::Entered<'a, P>>::block_on::h38c97d353b27ec23
    Jun 24 21:35:36 pi0 librespot[628]:   16:   0x4a85cb - <std::thread::local::LocalKey<T>>::with::h0f7467a351f98200
    Jun 24 21:35:36 pi0 librespot[628]:   17:   0x4a884f - <std::thread::local::LocalKey<T>>::with::h2b6e73dc3e24830a
    Jun 24 21:35:36 pi0 librespot[628]:   18:   0x4a8bcf - <std::thread::local::LocalKey<T>>::with::h515819cb127e3e09
    Jun 24 21:35:36 pi0 librespot[628]:   19:   0x4a47eb - tokio_core::reactor::Core::run::h43a6a16fa99f0069
    Jun 24 21:35:36 pi0 librespot[628]:   20:   0x4c77d3 - librespot::main::h583a074fae9ff638
    Jun 24 21:35:36 pi0 librespot[628]:   21:   0x4bacc3 - std::rt::lang_start::{{closure}}::h56d13da8f0c00177
    Jun 24 21:35:36 pi0 librespot[628]:   22:   0x97c477 - std::panicking::try::do_call::h803676a6686fa35a
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/rt.rs:59
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/panicking.rs:306
    Jun 24 21:35:36 pi0 librespot[628]:   23:   0x993943 - __rust_maybe_catch_panic
    Jun 24 21:35:36 pi0 librespot[628]:                        at libpanic_unwind/lib.rs:102
    Jun 24 21:35:36 pi0 librespot[628]:   24:   0x97990f - std::rt::lang_start_internal::h8d13ab8f0c59613e
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/panicking.rs:285
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/panic.rs:361
    Jun 24 21:35:36 pi0 librespot[628]:                        at libstd/rt.rs:58
    Jun 24 21:35:36 pi0 librespot[628]:   25:   0x4c8223 - main
    Jun 24 21:35:36 pi0 librespot[628]:   26: 0xb6d8e677 - __libc_start_main
    Jun 24 21:35:36 pi0 systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
    Jun 24 21:35:36 pi0 systemd[1]: raspotify.service: Unit entered failed state.
    Jun 24 21:35:36 pi0 systemd[1]: raspotify.service: Failed with result 'exit-code'.
    Jun 24 21:35:46 pi0 systemd[1]: raspotify.service: Service hold-off time over, scheduling restart.
    Jun 24 21:35:46 pi0 systemd[1]: Stopped Raspotify.
    Jun 24 21:35:46 pi0 systemd[1]: Starting Raspotify...
    Jun 24 21:35:46 pi0 systemd[1]: Started Raspotify.
    Jun 24 21:35:47 pi0 librespot[646]: INFO:librespot: librespot (raspotify v0.11.3) 431be9e (2018-05-18). Built on 2018-05-30. Build ID: GZE2acOG

tobiasmcnulty, 2018-06-24:
I take that back, it seems to be track dependent. It failed when I tried to select my output device immediately upon opening spotify, but when I started a track (playing locally) and then switched to my Raspberry Pi running raspotify, it worked okay. Apologies if this isn't any help (but do check out your log file(s) for any potential issues, as @stefjan suggested).

kennymc-c, 2018-06-25:
I had this problem with a previous and recent version of raspotify on my pi zero with stretch but there are no suspicious entries in my daemon.log. Also systemctl status raspotify shows no errors. After updating i could managed to get a connection but playing a song wasn't possible over the iOS App. A few tries later everything was like before the update and no connection was possible. I also tried the macOS App and numerous daemon restarts.
I'm also running shairport-sync for Airplay support on this device and there are no problems at all so wifi shouldn't be a problem here.

milyglos, 2018-06-25:
Hi! Thanks for the answers! I don't know what happened but yesterday it worked for couple of hours. Now I came back from work and just connected and the music is playing. I guess I have to use it more and see if it will be stable... I hope so :) 

But when I read the log from yesterday I see this happening in a loop for like 50minutes or so... WOW.

Jun 24 14:15:27 raspberrypi librespot[385]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-mp8m.ap.spotify.com:4070"
Jun 24 14:15:27 raspberrypi avahi-daemon[302]: Withdrawing address record for 192.168.0.248 on eth0.
Jun 24 14:15:27 raspberrypi avahi-daemon[302]: Host name conflict, retrying with raspberrypi-2
Jun 24 14:15:27 raspberrypi avahi-daemon[302]: Registering new address record for 2a02:a31a:e141:bc80:94ed:7419:401a:1306 on eth0.*.
Jun 24 14:15:27 raspberrypi avahi-daemon[302]: Registering new address record for 192.168.0.248 on eth0.IPv4.
Jun 24 14:15:27 raspberrypi librespot[385]: thread 'main' panicked at 'Authentication failed with reason: BadCredentials', core/src/connection/mod.rs:93:21
Jun 24 14:15:27 raspberrypi librespot[385]: stack backtrace:
Jun 24 14:15:27 raspberrypi librespot[385]:    0:   0x963c9b - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::hd7ee9e6e2fbd0546
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
Jun 24 14:15:27 raspberrypi librespot[385]:    1:   0x96ace3 - std::sys_common::backtrace::print::ha543f35261e05af8
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/sys_common/backtrace.rs:71
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/sys_common/backtrace.rs:59
Jun 24 14:15:27 raspberrypi librespot[385]:    2:   0x95a2a3 - std::panicking::default_hook::{{closure}}::h4e2726d8115dc6e7
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/panicking.rs:207
Jun 24 14:15:27 raspberrypi librespot[385]:    3:   0x959ef7 - std::panicking::default_hook::heb92eb7475fdb71c
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/panicking.rs:223
Jun 24 14:15:27 raspberrypi librespot[385]:    4:   0x95a74b - std::panicking::rust_panic_with_hook::hdc6f326636ed669e
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/panicking.rs:402
Jun 24 14:15:27 raspberrypi librespot[385]:    5:   0x95a567 - std::panicking::begin_panic_fmt::hc0834ed5723739e8
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/panicking.rs:349
Jun 24 14:15:27 raspberrypi librespot[385]:    6:   0x5d2d5b - <futures::future::chain::Chain<A, B, C>>::poll::h1148c6132503b5cf
Jun 24 14:15:27 raspberrypi librespot[385]:    7:   0x5f55b7 - <futures::future::and_then::AndThen<A, B, F> as futures::future::Future>::poll::h1cce02c614c2a85c
Jun 24 14:15:27 raspberrypi librespot[385]:    8:   0x5dde1f - <futures::future::chain::Chain<A, B, C>>::poll::h86ae1f1e239d835f
Jun 24 14:15:27 raspberrypi librespot[385]:    9:   0x62900b - <futures::future::map::Map<A, F> as futures::future::Future>::poll::h961e6c3a5beec363
Jun 24 14:15:27 raspberrypi librespot[385]:   10:   0x4a31bf - <librespot::Main as futures::future::Future>::poll::hc5558f1fdc6e23ed
Jun 24 14:15:27 raspberrypi librespot[385]:   11:   0x47c95f - <futures::task_impl::Spawn<T>>::poll_future_notify::hc9211ab06780cae1
Jun 24 14:15:27 raspberrypi librespot[385]:   12:   0x47c75f - <futures::task_impl::Spawn<T>>::poll_future_notify::ha8b42b6b522978eb
Jun 24 14:15:27 raspberrypi librespot[385]:   13:   0x48674f - <std::thread::local::LocalKey<T>>::with::h2c21813f4460189b
Jun 24 14:15:27 raspberrypi librespot[385]:   14:   0x49322f - <tokio::executor::current_thread::Entered<'a, P>>::block_on::hf43c8a2fdba49de7
Jun 24 14:15:27 raspberrypi librespot[385]:   15:   0x486947 - <std::thread::local::LocalKey<T>>::with::h76146f30987650d4
Jun 24 14:15:27 raspberrypi librespot[385]:   16:   0x486d6b - <std::thread::local::LocalKey<T>>::with::hd6791b0e6c65e9d1
Jun 24 14:15:27 raspberrypi librespot[385]:   17:   0x486617 - <std::thread::local::LocalKey<T>>::with::h2a5318e41afac7f6
Jun 24 14:15:27 raspberrypi librespot[385]:   18:   0x48280b - tokio_core::reactor::Core::run::ha49f6d3f8aa188c1
Jun 24 14:15:27 raspberrypi librespot[385]:   19:   0x4a57f3 - librespot::main::h54f3befa380b485c
Jun 24 14:15:27 raspberrypi librespot[385]:   20:   0x498ce3 - std::rt::lang_start::{{closure}}::h71794c74d2130e37
Jun 24 14:15:27 raspberrypi librespot[385]:   21:   0x95a3f7 - std::panicking::try::do_call::h5731926498d75d4e
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/rt.rs:59
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/panicking.rs:306
Jun 24 14:15:27 raspberrypi librespot[385]:   22:   0x9718c3 - __rust_maybe_catch_panic
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libpanic_unwind/lib.rs:102
Jun 24 14:15:27 raspberrypi librespot[385]:   23:   0x95788f - std::rt::lang_start_internal::h011c0cae803bbfbf
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/panicking.rs:285
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/panic.rs:361
Jun 24 14:15:27 raspberrypi librespot[385]:                        at libstd/rt.rs:58
Jun 24 14:15:27 raspberrypi librespot[385]:   24:   0x4a6243 - main
Jun 24 14:15:27 raspberrypi librespot[385]:   25: 0x76ce6677 - __libc_start_main
[... cut ...]
Jun 24 19:01:36 raspberrypi librespot[4420]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-v8h5.ap.spotify.com:4070"
Jun 24 19:01:37 raspberrypi librespot[4420]: INFO:librespot_core::session: Authenticated as "milyglos" !
...... I seriously put right username and password in conf ;) There is BadCredentials written what I understand like there was something wrong with my username and password right? And after 50 minutes of this loop it started to work... 

Any ideas? Now I am listening to 5th song in a row...

milyglos, 2018-06-26:
And this is what happened today. 

Jun 26 01:53:05 raspberrypi systemd[1]: Starting Daily apt download activities...
Jun 26 01:53:07 raspberrypi systemd[1]: Started Daily apt download activities.
Jun 26 01:53:07 raspberrypi systemd[1]: apt-daily.timer: Adding 5h 39min 25.730030s random time.
Jun 26 01:53:07 raspberrypi systemd[1]: apt-daily.timer: Adding 10h 18min 48.426391s random time.
Jun 26 06:51:40 raspberrypi librespot[4420]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-q7p6.ap.spotify.com:4070"
Jun 26 06:51:40 raspberrypi librespot[4420]: INFO:librespot_core::session: Authenticated as "milyglos" !
Jun 26 06:51:40 raspberrypi librespot[4420]: INFO:librespot_core::session: Country: "PL"
Jun 26 06:51:40 raspberrypi librespot[4420]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
Jun 26 06:51:41 raspberrypi librespot[4420]: INFO:librespot_playback::player: Loading track "missing" with Spotify URI "spotify:track:3jgx1CzgaUo9V3IJfzwtPW"
Jun 26 06:51:42 raspberrypi librespot[4420]: INFO:librespot_playback::player: Track "missing" loaded
Jun 26 06:52:59 raspberrypi librespot[4420]: INFO:librespot_playback::player: Loading track "By Chris Johnston, Craig Markva, Jamie Evans," with Spotify URI "spotify:track:30b6fM2TElAtXlnqZawe0l"
Jun 26 06:52:59 raspberrypi librespot[4420]: WARN:librespot_playback::player: Reducing normalisation factor to prevent clipping. Please add negative pregain to avoid.
Jun 26 06:53:01 raspberrypi librespot[4420]: INFO:librespot_playback::player: Track "By Chris Johnston, Craig Markva, Jamie Evans," loaded
Jun 26 06:57:17 raspberrypi systemd[1]: Starting Daily apt upgrade and clean activities...
Jun 26 06:57:21 raspberrypi systemd[1]: Started Daily apt upgrade and clean activities.
Jun 26 06:57:21 raspberrypi systemd[1]: apt-daily-upgrade.timer: Adding 50min 29.545811s random time.
Jun 26 06:57:21 raspberrypi systemd[1]: apt-daily-upgrade.timer: Adding 14min 9.024061s random time.
Jun 26 06:58:05 raspberrypi librespot[4420]: ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
Jun 26 07:08:16 raspberrypi librespot[4420]: ERROR:librespot_core::session: Os { code: 110, kind: TimedOut, message: "Connection timed out" }

And now I wanted to connect by android phone app but again, I have seen the raspotify on the list but couldn't connect. And now it's not listed on the phone anymore. This is the log from the time I wanted to connect by phone:

Jun 26 07:19:27 raspberrypi dbus[315]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service'
Jun 26 07:19:27 raspberrypi systemd[1]: Starting Hostname Service...
Jun 26 07:19:27 raspberrypi dbus-daemon[473]: Activating service name='ca.desrt.dconf'
Jun 26 07:19:27 raspberrypi dbus-daemon[473]: Successfully activated service 'ca.desrt.dconf'
Jun 26 07:19:28 raspberrypi dbus[315]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jun 26 07:19:28 raspberrypi systemd[1]: Started Hostname Service.

Like nothing from librespot...hmmmm....Oh. And it also disappeared from the desktop spotify device list.

dtcooper, 2018-07-11:
What tracks is it failing on?

milyglos, 2018-07-12:
Well generally as I observe the problem is that I come from work and try to connect and sometimes it connects sometimes not. I reboot the raspberry and after some time it shows up again in spotify list and I can or can't connect. I reboot again and finally it connects and I can listen to the music. I play it and do stuff and generally they are my playlists. While listening to the music I do other stuff so I don't know when it stops playing. I should maybe spaend a day home and test it but it is hard for me to organize such a time ;)

milyglos, 2018-07-12:
Ok. So I played couple of playlists and IT was ok. But on the third song of a radio based on one od my playlists IT disconnected and disappeared from the list...

konstantingoretzki, 2018-11-18:
> Hi Kamil
>
> i had a similar problem: raspotify showed up in the Spotify connect list, but i could not connect (connecting -> standby). The problem was that i had port 4070 blocked in my firewall. [...]

Thank you very much!

I had the same problem that I was not able to connect to my raspberry (click on connect inside e.g. the application on Windows and nothing happens) and this fixed it for me.
The alternative would be to enter your credentials inside the conf file `/etc/default/raspotify`.

lamoe02, 2019-01-17:
without any warranty but I think this resolved this issue i.e. opened up the port 4070:

 sudo iptables -I INPUT -p tcp --dport 4070 -j ACCEPT

S0AndS0, 2019-07-12:
Pretty sure that...

```
Jun 24 15:04:03 raspberrypi librespot[4420]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-xcjx.ap.spotify.com:4070"
```

... is connecting to a remote server on `4070` so doing as @lamoe02 suggests likely will have no effect, I think something like...

```
sudo iptables -I OUTPUT -p tcp --dport 4070 -j ACCEPT
```

... is what was intended, however, from my own tests it looks like there's two ports that are listening for Raspotify...

```Bash
netstat -plantu | grep -- '/librespot'
#> tcp         0      0 0.0.0.0:36843          0.0.0.0:*         LISTEN      1630/librespot
#> udp         0      0 0.0.0.0:5353           0.0.0.0:*                     1630/librespot
```

... of which only `5353` is consistent and the `tcp` port will change any time the service is restarted; so for services like Raspotify I've a Bash function that goes something like...

[**`await-service-port.sh`**](https://github.com/bash-utilities/await-service-port/blob/master/await-service-port.sh)

```Bash
#!/usr/bin/env bash

## Lists listening ports for given service and protocol via `netstat`
## @returns {number|list}
## @param {string} _service_name      - Name of service to `grep` for
## @param {string[tcp|udp]} _protocol - Protocol that service communicates through
## @param {number} _sleep_interval    - Number of seconds to sleep between checks
## @param {number} _loop_limit        - Max number of loops before function returns error code
## @author S0AndS0
## @copyright AGPL-3.0
## @exampe As an array
##     _sshd_ports_list=($(await_service_port 'sshd'))
##     printf 'SSHD listening port: %s\n' "${_sshd_ports_list[@]}"
##     #> SSHD Port: 22
##     #> SSHD Port: 2222
## @example As a string
##     _sshd_ports_string="$(await_service_port 'sshd')"
##     printf 'SSHD listening port(s): %s\n' "${_sshd_ports_string}"
##     #> SSH Ports: 22 2222
await_service_port(){
    local _service_name="${1:?# Parameter_Error: no service name supplied to ${FUNCNAME}}"
    local _protocol="${2:-tcp}"
    local _sleep_interval="${3:-1}"
    local _loop_limit="${4:-10}"

    if [ "${_sleep_interval}" -le '0' ] || [ "${_loop_limit}" -le '0' ]; then
        printf 'Parameter_Error: %s requires positive numbers for third and forth parameters\n' "${FUNCNAME[0]}" >&2
        return 1
    fi

    case "${_protocol,,}" in
        tcp)
            local _netstat_opts='plant'
        ;;
        udp)
            local _netstat_opts='planu'
        ;;
        *)
            printf '# Parameter_Error: protocol must either be "tcp" or "udp"' >&2
            return 1
        ;;
    esac

    local _loop_count='0'
    local -a _listen_ports
    while true; do
        local _netstat_service="$(netstat -${_netstat_opts} 2>/dev/null | grep -E -- "/${_service_name}( +)")"
        for _port in $(awk '{gsub(":"," "); print $5}' <<<"${_netstat_service}"); do
            _listen_ports+=("${_port}")
        done

        if [ "${#_listen_ports[@]}" -gt '0' ]; then
            printf '%s\n' "${_listen_ports[*]}"
            break
        elif [ "${_loop_count}" -gt "${_loop_limit}" ]; then
            break
        fi

        let _loop_count+='1'
        sleep "${_sleep_interval}"
    done

    [[ "${#_ipv4_addresses[@]}" -gt '0' ]]; return "${?}"
}
```

... that lists listening ports, even those dynamically assigned at service start, which may be utilized in scripts kind like...

**`example-librespot-iptables.sh`**

```Bash
#!/usr/bin/env bash

## Find true directory this script resides in
__SOURCE__="${BASH_SOURCE[0]}"
while [[ -h "${__SOURCE__}" ]]; do
    __SOURCE__="$(find "${__SOURCE__}" -type l -ls | sed -n 's@^.* -> \(.*\)@\1@p')"
done
__DIR__="$(cd -P "$(dirname "${__SOURCE__}")" && pwd)"
__NAME__="${__SOURCE__##*/}"

## Source module code within this script
source "${__DIR__}/modules/await-service-port/await-service-port.sh"

start_up(){
    iptables --new-chain eth0_input_raspotify
    iptables --new-chain eth0_output_raspotify

    _raspotify_tcp_ports=($(await_service_port 'librespot' 'tcp' '1' '19'))
    for _port in "${_raspotify_tcp_ports[@]}"; do
        iptables -A eth0_input_raspotify\
         -i eth0\
         -s 192.168.0.1/24\
         --dport ${_port}\
         -p tcp\
         -m conntrack --ctstate NEW,ESTABLISHED -m limit --limit 5/second --limit-burst 100 -j ACCEPT

        iptables -A eth0_output_raspotify\
         -o eth0\
         -d 192.168.0.1/24\
         --sport ${_port}\
         -p tcp\
         -m conntrack --ctstate ESTABLISHED -j ACCEPT
    done

    eth0_input_raspotify
}

stop_down(){
    iptables -F eth0_output_raspotify
    iptables -X eth0_output_raspotify

    iptables -F eth0_input_raspotify
    iptables -X eth0_input_raspotify
}

## Parse the start/stop argument
case "${1,,}" in
    'start'|'up')
        start_up
    ;;
    'stop'|'down')
        stop_down
    ;;
    *)
        printf '%s cannot do that right now' "${__NAME__}" >&2
        exit 1
    ;;
esac
```

... although above's incomplete it _should_ provide a starting point for those that don't want to await other related projects to be _`push`ed_...

> Hint, I utilize `systemd` to tell the `example-librespot-iptables.sh` script what to do; it's only started when both a select interface and the service are ready, and stopped when either the service or interface are taken down.

... as it holds the _morals_ of what iptables rules are required to have happy clients when Raspotify is on a device with the default output policy set to `REJECT` and input policy set to `DROP`

CoLuxe, 2021-07-27:
I had the same problem, used google, found the thread, found a better solution and want to write it here for other guys:

For this problem there is a much simpler solution.
In Librespot exists an option for this, see here: https://github.com/librespot-org/librespot/wiki/Options
Just use "--zeroconf-port <port>" to set a static port.

closed 2021-11-24 by JasonLG1979