dtcooper/raspotify #127: Raspotify not working on USB Audio output and Alientek D8 amplifier

HexaSky, 2018-06-28:
Hi guys,

Yesterday I've tried to installed Raspbian and Raspotify on my RPI 3. It worked very well using HDMI output. 
Today I want to connect directly my Raspberry Pi on my Alientek D8 amplifier via USB output of the Raspberry. My smartphone still detect raspotify but I've no sound via USB, only work using HDMI output. 

When I look at audio settings in Raspbian, I can see that my Alientek D8 is in the peripherals list but can't have any sound. I've selected my amplifier to be my default audio device but don't work. I also see that when I right click on the speaker icon, the HDMI label is checked and when I want to select my Alientek, it just don't work at all. The check mark still on HDMI. 

Have you an idea of what I can do to make my system work on Raspberry Pi audio output ?

PS : the USB chip of my amplifier is a XMOS chip.

Thank you for your help in advance 😄

HexaSky, 2018-07-02:
Up !

Nobody have an answer ? I've have done many tests but always nothing...

dtcooper, 2018-07-03:
Hi @HexaSky, try messing around with the settings file @ `/etc/default/raspotify` and add a `--device` argument to `OPTIONS`. You can google around how to get the hardware ID of the USB device, or you can just try `--device hw:0,1`.

Look for,

```
# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username <USERNAME> --password <PASSWORD>"
```

HexaSky, 2018-07-03:
Hi David, thank you for your answer !

I’ve just found a solution before you send me that message ! It know works well. The only last problem is that little cut or crack sound appear sometimes. Can’t find any fix for this problem…

dtcooper, 2018-07-03:
How did you end up solving this? I'd be happy to add it to the [_Troubleshooting_](https://github.com/dtcooper/raspotify#troubleshooting) section of the Readme to help others.

franz159, 2018-09-02:
I have an at all similar problem with my Chord MoJo attached via USB on top of Moode 4.2

If I use my IQaudIO+ Hat Dac, directly connected to my pi raspotify works.
But if I set up the USB device I have no sound.
Looking at /var/log/syslog everything seems fine (spotify session ok, loading track ok...etc) but I have no sound.
Since my device is seen as 
pi@moode:/var/log $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
**card 1: Mojo [Mojo], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0**

I tried adding the "--device hw:1,0" option but still I have no sound

franz159, 2018-09-02:
Not sure if this is relevant at all, but to make Chord MoJo work with Airplay, I had to set the bit depth to 32 in shairport config, but I cannot find anything similar in raspotify

closed 2021-11-24 by github-actions[bot]