dtcooper/raspotify #128: Output hdmi works when the av receiver is turned on

foncubierta, 2018-06-28:
I have a raspberry pi 3 that works perfect if it starts and the av receiver is on (brand onkyo connected by hdmi)
If I turn on the raspberry pi and it's off, the av receiver does not work.
Do you have any ideas because it can be?

# /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
#DEVICE_NAME="raspotify"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="320"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username <USERNAME> --password <PASSWORD>"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=10"

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"

mini81, 2018-07-08:
Same here! Onkyo Receiver and Pi3.

vguttmann, 2020-07-17:
If the issue is resolved, you should consider closing it @foncubierta

closed 2021-11-24 by github-actions[bot]