dtcooper/raspotify #129: [RPi3] slower song speed

zamszowy, 2018-07-08:
Hi,

I've noticed that song played on Raspberry Pi 3 (with OSMC 2018.06-1 installed) has significantly slower speed than played on other platforms (phone, desktop computer).
You can see the difference in the two recordings (MP3 format) that I've made: [slow-speed.zip](https://github.com/dtcooper/raspotify/files/2173652/slow-speed.zip)

Does anyone know how to fix this?

zamszowy, 2018-07-10:
Seems that this is related to OSMC itself, as I've just installed raspbian and the issue does not exist there (also the quality is better).

steinhmor, 2018-09-16:
I'm having similar problems with raspotify on dietpi. Maybe it's not OSMC related?

zamszowy, 2018-09-24:
> @zamszowy can you please reopen this issue?

Done.

closed 2021-11-24 by github-actions[bot]