dtcooper/raspotify #130: Stops Playing - Pulse Audio Error

Terranigmus, 2018-07-08:
pi@raspberrypi:~ $ sudo systemctl status raspotify
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: active (running) since So 2018-07-08 12:04:57 UTC; 14min ago
  Process: 1367 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1364 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1370 (librespot)
   CGroup: /system.slice/raspotify.service
           ├─1370 /usr/bin/librespot --name raspotify (raspberrypi) --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --username terranigmus --password garbl123
           └─1489 /usr/bin/pulseaudio --start --log-target=syslog

Jul 08 12:10:16 raspberrypi librespot[1370]: INFO:librespot_playback::player: Track "Me and My Friends" loaded
Jul 08 12:10:17 raspberrypi pulseaudio[1489]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jul 08 12:10:17 raspberrypi pulseaudio[1489]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
**Jul 08 12:10:17 raspberrypi pulseaudio[1489]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org....er" doesn't exist**
Jul 08 12:10:39 raspberrypi librespot[1370]: INFO:librespot_playback::player: Loading track "Numb" with Spotify URI "spotify:track:2woHPlGnIrxBWbDO4I2UOb"
Jul 08 12:10:43 raspberrypi librespot[1370]: INFO:librespot_playback::player: Track "Numb" loaded
Jul 08 12:14:03 raspberrypi librespot[1370]: INFO:librespot_playback::player: Loading track "It Will Stand" with Spotify URI "spotify:track:75jmjhxqw7v31AevBDTrJw"
Jul 08 12:14:06 raspberrypi librespot[1370]: INFO:librespot_playback::player: Track "It Will Stand" loaded
Jul 08 12:17:36 raspberrypi librespot[1370]: INFO:librespot_playback::player: Loading track "The Last Of The Real Ones (Remix) [feat. MadeinTYO & bülow] - Remix" with Spotify URI "spotify:track:0INAgsMJSIwCcWTgIuyaMY"
Jul 08 12:17:38 raspberrypi librespot[1370]: INFO:librespot_playback::player: Track "The Last Of The Real Ones (Remix) [feat. MadeinTYO & bülow] - Remix" loaded
Hint: Some lines were ellipsized, use -l to show in full.

Basically the old "stops after a few songs", just a different error message this time

zamszowy, 2018-07-08:
These logs contain your username and password, which probably wasn't something you intended to share here (if these are real ones).

JasonLG1979, 2021-11-17:
Raspotify is primarily intended to be used in a headless enviroment (Raspberry Pi OS lite) For desktop versions of Raspberry Pi OS (and/or systems with PulseAudio installed) [spotifyd](https://spotifyd.github.io/spotifyd/installation/Raspberry-Pi.html) offers similar functionality, much better PulseAudio compatibility and is a better choice.

closed 2021-11-17 by JasonLG1979