dtcooper/raspotify #131: no sound with hifiberry dac pro+ and out rca

Narmo70, 2018-07-11:
Hi everybody,
Sorry if I'm not in the right place but I need help!
I have a raspberry 3 plugged with a hifiberry dac pro +.
Everything works fine I can listen to Spotify (premium account) since volumio but I would like to use raspotify to listen to the music from the Spotify apk.

I did some research and I get to have the choice in the spotify apk to have the sound on my raspberry but the sound does not come out in rca since the hifiberry but only in hdmi?

could you help me?

A big thank you in advance

Narmo70, 2018-07-15:
I come back to you to find out if you had an idea about my request or if I'm not specific enough.

gfro84, 2018-07-15:
Google search for 'blacklist rapi hdmi sound'. Makes the berry the default and disables the pi's onboard sound, which I'm guessing won't be a loss.

vorbiss, 2018-10-08:
Yop I use  hifiberry dac + jack, it works well. I'll send you my conf, maybe it gonna help u.

See my project (french) : https://forum.hardware.fr/hfr/OverclockingCoolingModding/Divers-8/raspberry-meuble-connecte-sujet_285963_1.htm

vorbiss, 2018-10-08:
My conf

`# Device name on Spotify Connect
DEVICE_NAME="RaspiRadio"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="320"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info.
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
OPTIONS="--username xxxxxx --password xxxxxx --device plughw:CARD=sndrpihifiberry"

# Uncomment to use a cache for downloaded audio files. It's best to leave this
# as-is since permissions are properly set on directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"
~
`

3579km, 2018-11-30:
Try raspbian image with HiFiBerry auto-detection https://www.hifiberry.com/build/download/ and install Raspotify on it

closed 2021-11-24 by github-actions[bot]