dtcooper/raspotify #132: Can't play via bluetooth speakers

jkoenig72, 2018-07-11:
Hi,

quite new to all this Bluetooth connectivity on raspi.

However I get my Echo Dot setup with my raspi, so I can play a wav with aplay.

But how I get raspotify to play via this bluetooth connection?

I guess its named default, here is my .asoundrc

pcm.!default {
        type plug
        slave.pcm {
                type bluealsa
                device "FC:A6:67:3F:74:35"
                profile "a2dp"
        }
}

ctl.!default {
        type bluealsa
}

What I tried so far is play around with /etc/default/raspotify

OPTIONS="--device default"

and / or

BACKEND_ARGS="--backend alsa"

Still not working. Again BT is working, I can play both mp3 / wav but raspotify is simply not using that audio channel.

Thanks!
Joerg

jkoenig72, 2018-07-11:
When I do aplay -L this looks strange to me ... any idea?

pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default

jkoenig72, 2018-07-11:
When I stop raspotify service and start 

librespot --name "librespot test" --backend alsa --cache /tmp/librespot-test --bitrate 160 -v

I got those lines...

TRACE:librespot_audio::fetch: requesting chunk 32
TRACE:librespot_audio::fetch: chunk 32 / 33 complete
DEBUG:librespot_audio::fetch: File 2c319df6c39f009157205a7595e281ffb75bb4d7 complete, saving to cache
DEBUG:librespot_connect::spirc: kMessageTypePlay "JOERG\'s iPhone" 0accb07b5ec309f4819d0ab9688ccd970d934aa5 3 0
DEBUG:librespot_playback::player: command=Play
ALSA lib pcm.c:8481:(snd_pcm_set_params) Unable to set hw params for PLAYBACK: Input/output error
ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open -5
ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
ALSA lib pcm.c:8481:(snd_pcm_set_params) Unable to set hw params for PLAYBACK: Input/output error
ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open -5
ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
ALSA lib pcm.c:8481:(snd_pcm_set_params) Unable to set hw params for PLAYBACK: Input/output error
ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open -5
ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
ALSA lib pcm.c:8481:(snd_pcm_set_params) Unable to set hw params for PLAYBACK: Input/output error
ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open -5

jkoenig72, 2018-07-11:
Even this works - so I guess its a minor thing...

aplay -D bluealsa:HCI=hci0,DEV=FC:A6:67:3F:74:35,PROFILE=a2dp  c304-2.wav

jkoenig72, 2018-07-11:
Got it working by adding

librespot --name raspberry --bitrate 320 --backend alsa --device default

key seems "--device default"

jkoenig72, 2018-07-11:
to make it run with raspotify I needed to copy my .asoundrc to /etc/asound.conf 

I also think the raspotify.service and they it passes config values to librespot - it works for me if I hardcoded them....

pi@raspberrypi:~ $ cat /etc/systemd/system/multi-user.target.wants/raspotify.service
[Unit]
Description=Raspotify
After=network.target

[Service]
User=raspotify
Group=raspotify
Restart=always
RestartSec=10
PermissionsStartOnly=true
ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify ; /bin/chown raspotify:raspotify /var/cache/raspotify
Environment="DEVICE_NAME=Bad Joerg"
Environment="BITRATE=160"
Environment="CACHE_ARGS=--disable-audio-cache"
Environment="VOLUME_ARGS=--enable-volume-normalisation --linear-volume --initial-volume=100"
Environment="BACKEND_ARGS=--backend alsa"
EnvironmentFile=-/etc/default/raspotify
#ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS --device default
**ExecStart=/usr/bin/librespot --name "Bad Jörg" --bitrate 320 --backend alsa --device default**

[Install]
WantedBy=multi-user.target

DanielSSilva, 2018-07-23:
Hey @jkoenig72  thank you for this, it really helped me! All that you say is correct, but I've found that you can get that result without "harcoding" the configs. If you use the [/etc/default/raspotify](https://github.com/dtcooper/raspotify#configuration) , uncomment the `#BACKEND_ARGS="--backend alsa"` and set `OPTIONS="--device default"` it also works :)

skywinder, 2020-08-29:
there is 2 solutions:
1) https://github.com/dtcooper/raspotify/issues/63#issuecomment-411207480
2) https://github.com/dtcooper/raspotify/issues/172#issuecomment-605462378

skywinder, 2020-10-19:
@dtcooper this is also should be solved as described in #303

closed 2021-11-24 by github-actions[bot]