dtcooper/raspotify #133: Spotify connects but play button doesn't work

ematzdor, 2018-07-12:
Unfortunately, since this is normally when I'd describe expected behavior, I don't know what should happen. I just installed and it doesn't work.
I expected to be able to connect, hit play, and the song would play.

What actually happened is I would connect and hit play on a song, and it would flicker to pause then back to play. No sound plays on the Pi.

I don't know what I expect the service status to show, but it says stuff about panicking once I attempt to connect my phone. (The top run is just when the service started, the bottom is after connecting my phone and trying to play.)
https://etekweb.net/raspotify.jpg

Any help or guidance would be appreciated. Tried searching for the error but there wasn't anything that was similar to what I'm getting. Thanks.

SopsoN, 2018-07-16:
I have same issue. Sometimes it starts play song, but after 2 or 3 songs it is stopping by itself.

bpieres, 2018-07-28:
same issue.

closed 2021-11-24 by github-actions[bot]