dtcooper/raspotify #135: How to start raspotify when no systemctl is available

besi, 2018-07-22:
After the installation of raspotify I get the following error:

```
The user `raspotify' is already a member of `audio'.                                 /var/lib/dpkg/info/raspotify.postinst: 22: /var/lib/dpkg/info/raspotify.postinst: deb-systemd-helper: not found                                                           /var/lib/dpkg/info/raspotify.postinst: 25: /var/lib/dpkg/info/raspotify.postinst: deb-systemd-helper: not found                                                           /var/lib/dpkg/info/raspotify.postinst: 32: /var/lib/dpkg/info/raspotify.postinst: deb-systemd-helper: not found                                                           root@phone2:~# uname -a                                                              Linux phone2 3.18.0-trunk-rpi #1 PREEMPT Debian 3.18.5-1~exp1+rpi19 (2015-08-08) armv6l GNU/Linux                                                                         
```
Is there a way how I could (manually) startthe service.

I did a search for a binary but only found the following folders and files:

```
root@phone2:~# find / -name raspotify                                                /var/cache/raspotify                                                                 /usr/share/doc/raspotify                                                             /etc/default/raspotify                                                               root@phone2:~#                                                                            
```

michalfita, 2018-07-23:
I'd suggest to start with:
```bash
librespot --help
```

besi, 2018-07-23:
Indeed this does the trick. Thanks very much!

    Librespot —name SpotifyDevice

closed 2018-07-23 by besi