dtcooper/raspotify #138: Pi Zero W - wont connect

Locho24312, 2018-07-25:
Software:
Latest Raspbian
Latest Raspotify

Hardware:
Pi Zero W
Pimoroni - Phat Dac

Variables:
Wifi connected.

Problem:
I can, (play a test output and another audio file), however, the Pi will show up in 'spotify connect', cannot connect, and will either keep popping up in the connect devices tab or just dissapear all together, seems the pi once connection attempt happens it then dissapears. A reinstall will bring it back, which it will do the same thing(see note).
Could this be a port issue for the Pi Zero W??

I have also tried both configs of having default and PW and User in the config file.

Note:
I got this to work 1 time, where i could disconnect from the Pi, reconnecting was fine. When i;
 $ sudo reboot
The device then exhibited the above behaviour.

dtcooper, 2018-07-26:
Thanks for the report! You've given me just the excuse I need to purchase a Pi Zero W. :)

dtcooper, 2018-07-26:
If I don't update in a week, please bug me.

Locho24312, 2018-07-26:
Sounds Great!, let me know if you need anything in the mean time. Im going to keep trying a few more things and see if we can ratify it!

Locho24312, 2018-08-22:
@dtcooper How did you go? :)

pyrosmiley, 2018-09-04:
I’m having a similar issue, but the weird thing is that it used to work just fine! I have this and Shairport-sync running. Both used to work with few to no issues (save for a crackling sound on shairport that I resolved), and then suddenly — boom — can’t connect to raspotify. No changes to network, hardware, software, anything. 

It’ll show up as a device and show as connected until I try to play something, then that won’t work and it’ll vanish until I force quit and re-open Spotify. 

Also on a Pi Zero W.

LeoTronick, 2018-09-13:
Hi all,

Just to let you know. I installed raspotify on my pi zero w (set up with NOOBS just minutes before) , adapted the config and restarted service. I was immediately able to connect from my phone to the raspotify client and play Musik over hdmi port.
Even after a reboot it works nicely.
What I am still struggling with is change the config to play through a USB sound card. The card itself works right out of the box with pi zero w. I just forgot how to figure out the hw address of the device. Would someone have a hint on that?

Cheers 
Leo

dtcooper, 2018-11-11:
Thanks @Dennis14e -- could you submit a PR that adds this to the [Troubleshooting on the Readme](https://github.com/dtcooper/raspotify#troubleshooting)? If not, I'll get to it soon.

Closing now that we have a solution.

closed 2018-11-11 by dtcooper

jgabriel98, 2020-07-25:
> Hi all,
>
> Just to let you know. I installed raspotify on my pi zero w (set up with NOOBS just minutes before) , adapted the config and restarted service. I was immediately able to connect from my [...]

I have this exactly same issue.

Raspotify + shairport, when sudenttly raspotify wont connect anymore.

did you solve this?