dtcooper/raspotify #139: Raspotify and uff firewall?

tobbe111, 2018-08-09:
Dear all,

I just installed raspotify on my pi (latest stretch installed), and it works like a charm - thanks for developing it!

I also wanted to active ufw as a firewall, but found that I cannot have both on: when ufw is enabled, the pi does not show up on spotify connect anymore (and if I turn it on in the middle of a song playing on my pi, I lose control over it and cannot, e.g., pause the song anymore). So far so good (and somewhat obvious). Is there a simple way to have both ufw and raspotify coexist, e.g. by opening a port or so? Thanks so much in advance!

dtcooper, 2018-09-06:
This would be ufw firewall configuration issue. Check out the `--ap-port` arg for librespot as well. Closing.

closed 2018-09-06 by dtcooper

Navneet-Suresh, 2020-02-08:
I had the same issue!
I just added zeroconf-port into the Raspotify config and then whitelisted the ports used by librespot.
`OPTIONS="--zeroconf-port 65444"`
I found out the ports by using netstat.
`sudo netstat -tulpn | grep libre`

Navneet-Suresh, 2020-02-08:
Download the app profile from here and remove .txt extension.

[raspotify.txt](https://github.com/dtcooper/raspotify/files/4174456/raspotify.txt)

UFW app profile can be placed at /etc/ufw/applications.d, After you add it reload the apps
```
ufw app update raspotify
ufw app info raspotify
ufw allow raspotify
```

rajanski, 2020-11-15:
> Download the app profile from here and remove .txt extension.
>
> [raspotify.txt](https://github.com/dtcooper/raspotify/files/4174456/raspotify.txt)
>
> UFW app profile can be placed at [...]
Great!
I actually had to change the port line in the ufw app like that `ports=5353/udp|65444/tcp to make it work
`

Navneet-Suresh, 2020-11-16:
@rajanski glad that it worked! noted. I switched to Amazon Prime/YouTube Premiuim combo as I got a better deal on them.
I don't use Spotify anymore and by extension Raspotify.

lfg6000, 2021-05-31:
here what i did to get raspotify to work with ufw enabled
(many thanks to  meta-n4vn33t and rajanski for their very helpful posts)

Edit raspotify config file
**> sudo nano /etc/default/raspotify                   
   // - edit this config file to change device name displayed in spotify app and  
   // - add a OPTIONS line so raspotify always uses port 65444 otherwise the port changes on every boot
   //    which makes setting a ufw rules of raspotify impossible
   DEVICE_NAME="LivingRoom"      
   OPTIONS="--zeroconf-port 65444" 

UFW and raspotify
paste in these rules into a terminal (the ip addresses may need to be updated to match your network) 
**> sudo ufw allow from 192.168.0.0/16 to any port 22 proto tcp
**> sudo ufw allow from 192.168.0.0/16 to any port 5900
**> sudo ufw allow from 192.168.0.0/16 to any port 5353 proto udp
**> sudo ufw allow from 192.168.0.0/16 to any port 65444 proto tcp
**> sudo ufw enable
**> reboot
        port descriptions
           ssh (22) (tcp)                   - winscp
           vnc (5900) (tcp) (udp)     - vnc server
           raspotify (65444 tcp)      - spotify connect
           raspotify (5353 udp)       - spotify connect