dtcooper/raspotify #141: Controlling playback without Web API

georgefst, 2018-08-23:
This may be a stupid question, but is there any possibility of being able to play/pause tracks etc. through the command line or through a local program? I have a script set up on my Pi which uses [spotipy](https://github.com/plamere/spotipy/blob/master/docs/index.rst) to control playback, but it seems a slightly convoluted route. Plus, there's a noticeable lag. It would make sense to communicate with raspotify directly.

cadavre, 2018-09-21:
@georgefst could you share your `spotipy` code? I'm looking for some (any, actually) simple way to call play/pause, next/prev track to Spotify and I cannot find anything. :/

Adding playback API would mean creating another interface.

georgefst, 2018-09-27:
Hi there, sorry this has taken me a while. I can sympathise as it took me a while to work out a good workflow for spotipy. In fact, when I've got more time on my hands, I'm gonna see about forking/contributing with an addition to the API to automate some of this stuff.

This code is stripped out of a larger script for controlling things with buttons on a rPi. I think it's mostly pretty self-explanatory but ask me if you have any questions. The only remotely clever part is that it wraps everything in a `spotifyAction` which refreshes the access token when necessary.

There are some constants which need to be entered:
- The `client_id` and `client_secret` come from the spotify developer dashboard.
- The `refreshToken` I think comes from running `spotipy.util.prompt_for_user_token`. I can't remember the exact process.
- The `device_id` might not be necessary for you, but you can find it by running `spotify.devices()` (after setting up the `spotify` object by running `refreshAccessToken`).

You can then call any of the functions at the end e.g. `playPause`.

```python
import spotipy
import spotipy.oauth2

# spotify controls setup
pi_device_id = ''
my_client_id = ''
my_client_secret = ''
my_redirect_uri = 'https://www.last.fm/user/george_____t/library/artists'
my_scope = 'streaming user-read-currently-playing user-read-playback-state'
refreshToken = ''
auth = spotipy.oauth2.SpotifyOAuth(client_id=my_client_id, client_secret=my_client_secret, redirect_uri=my_redirect_uri, scope=my_scope)

def refreshAccessToken():
  global accessTokenInfo, accessToken, spotify
  accessTokenInfo = auth.refresh_access_token(refreshToken)
  accessToken = accessTokenInfo['access_token']
  spotify = spotipy.Spotify(accessToken)

refreshAccessToken()

# wrapper for refreshing token - should make sure 'action' can be run partially then fully without any issues
def spotifyAction(action, description):
  try: # in case of failure (eg. wifi off, no song playing), don't do anything
    action()
  except Exception as e:
    print('Failed to ' + description + ': ' + str(e))
    if spotipy.oauth2.is_token_expired(accessTokenInfo):
      refreshAccessToken()
      spotifyAction(action, description)

def playPauseAction():
  isOnPi = (d for d in spotify.devices()['devices'] if d['id'] == pi_device_id).next()['is_active']    
  if not isOnPi:
    spotify.transfer_playback(pi_device_id)
  else:
    isPlaying = spotify.current_user_playing_track()[u'is_playing']
    spotify.pause_playback() if isPlaying else spotify.start_playback()

playPause = lambda: spotifyAction(playPauseAction, 'play/pause')
nextTrack = lambda: spotifyAction(spotify.next_track, 'go to next track')
prevTrack = lambda: spotifyAction(spotify.previous_track, 'go to previous track')
impossibleSoul = lambda: spotifyAction(seqComp([lambda: spotify.start_playback(uris=['spotify:track:5CLs0uFRmU0U9VcnsI6jwv']), lambda: spotify.seek_track(765000), lambda: spotify.transfer_playback(pi_device_id)]), 'get hype')
```

cadavre, 2018-09-28:
@georgefst marvelous! Code is pretty simple, I'm barely jr in Python development and also never used spotipy. In your snippet I can find literally *all* actions I had on mind to accomplish what I need.

I also think implementing this into Raspotify shouldn't be huge problem, except it will possibly require running HTTP server to handle some Request - Response between system and Spotify's OAuth.

Aranos49, 2018-10-12:
This code is very interesting. Have you tried it ?
Must it be added to the raspotify config file ?

georgefst, 2018-10-15:
> This code is very interesting. Have you tried it ?

Yes, it's my code, I utilise it pretty much every day.

> Must it be added to the raspotify config file ?

No, it's a standalone Python script.

georgefst, 2021-11-17:
FWIW, `spotifyd`, which I switched to *long* ago, allows local controls via e.g. `dbus-send --print-reply --dest=org.mpris.MediaPlayer2.spotifyd /org/mpris/MediaPlayer2 org.mpris.MediaPlayer2.Player.PlayPause`. `PlayPause` can be replaced with `Play`, `Pause`, `Next`, `Previous`...

closed 2021-11-17 by JasonLG1979