dtcooper/raspotify #144: Limit cache size

nicolasbrailo, 2018-09-04:
Raspotify started failing today (playing a few seconds, then disconnecting). I noticed /var/cache/raspotify was taking up 14 of the 15gb in the storage. Clearing the cache solved the issue.

Can the cache be limited in size through a configuration option? It might also be good to fail starting the server if there's not enough disk space.

laurensnl, 2018-12-27:
Good idea. Had the same problem here.

dtcooper, 2018-12-27:
Let me look into this. It could be a librespot problem or just the daemon not cleaning up.

chrstphrknwtn, 2019-01-07:
I had the same issue (playback starts then immediately stops because the disk is full with cached tracks) I assume `librespot` attempts to write to disk and fails, then just gives up.

The ultimate solution would be to have an auto-purging cache, where some combination of time since last play and play frequency is used to determine tracks to purge from the cache.

JonnieCache, 2019-06-11:
chalk me up as another one with this issue

AikedeJongste, 2019-09-06:
Yes, same here. A workaround is to set the cache directory to /tmp/ in /etc/default/raspotify, because /tmp will be wiped after a reboot. 

`CACHE_ARGS="--cache /tmp/raspotify"
`

Nosskirneh, 2019-11-03:
Any news on this? The provided workaround above isn't ideal as my Connect pi is never rebooted. I'm thinking about doing a cron job to wipe the cache once it becomes too full.

dtcooper, 2019-11-03:
If you can write daily cron job that deletes from the folder... that may help. I'll work on that if you can't get to it and include it in the package.

x0rtrunks, 2020-03-08:
From what I can gather librespot has the option to disable the audio file cache at runtime with --disable-audio-cache, but how do I do that with raspotify?

x0rtrunks, 2020-03-08:
Cronjob it is then. For anyone else wondering how or what. this is the cronjob I used.

sudo crontab -e

then place this in your crontab to delete the cache files every 12 hour.

* */12 * * * sudo rm -rf /var/cache/raspotify/files/* >/dev/null 2>&1

If you need more or less requent and have no idea what the cron syntax is, just use crontab-generator.org and use the same sudo rm -rf /var/cache/raspotify/files/* command but pick a different time range.

frode, 2020-03-11:
> Cronjob it is then. For anyone else wondering how or what. this is the cronjob I used.
>
> sudo crontab -e
>
> then place this in your crontab to delete the cache files every 12 hour.
>
>     * _/12 * * * [...]

Thanks @x0rtrunks!

Working cron job (in code tag)

`*/12 * * * sudo rm -rf /var/cache/raspotify/files/ >/dev/null 2>&1`

jimjimjimjimjimjimjim, 2020-04-17:
With added check if size is above 8GB and deletion of oldest files (not really tested all in production yet, only in small test folder but worked in small test):

sudo nano cache_limiter.sh paste below and update crontab as comment

# /etc/crontab do this every 12 hours
# 0 */12 * * * sudo cache_limiter.sh

# folder to check
FOLDER='/var/cache/raspotify/files/'

# 8GB-ish = 8123456789
MAX_SIZE="8123456789"

check_size () {
  # get size of cache folder
  CHECK=$(du -sb $FOLDER | cut -f1)

#  echo "checking $CHECK larger than $MAX_SIZE?"
  # if folder size is greater that wanted
  if [ "$CHECK" -gt "$MAX_SIZE" ]; then
    echo 1
  else
    echo 0
  fi
}

# if folder size is greater that wanted
while [ "$(check_size)" -gt 0 ]
do
  # delete oldest file (ls -t $FOLDER | tail -1)
  OLDEST=$(ls -t $FOLDER | tail -1)
  sudo rm -rf $OLDEST >/dev/null 2>&1
  # and repeat until size is OK
done

khink, 2020-09-09:
Thanks for the script!
I found i had to replace `$OLDEST` (in `sudo rm -rf $OLDEST >/dev/null 2>&1`) with `$FOLDER/$OLDEST`. Without that, the script can't find the folder to delete, which goes unnoticed because output is sent to `/dev/null`, so we have a loop that doesn't end in the normal way.

shawaj, 2021-01-04:
+1 on this! Should be a flag to limit the size IMO

I also added a issue on librespot here https://github.com/librespot-org/librespot/issues/558

Johannesd3, 2021-05-08:
FYI librespot 0.2.0 has a flag `--cache-size-limit`

khink, 2021-05-08:
@Johannesd3 Thanks for fixing this! (For reference: https://github.com/librespot-org/librespot/pull/675)

closed 2021-11-24 by JasonLG1979