dtcooper/raspotify #145: Arch ARM compatibility

dusatvoj, 2018-09-07:
I'm missing compatibility with archlinuxarm.org

110101011, 2018-10-17:
Would be that hard, it just needs a line change in the source code.

Remove line 31 to 33 in install.sh and test out for yourself, it should at least install.

if lsb_release -si | fgrep -ivq raspbian; then
    run_on_pi_only
fi

dusatvoj, 2018-11-08:
@110101011 And what about APT? Arch is not using APT

Lerentis, 2018-11-11:
@dusatvoj you could use [debtap](https://github.com/helixarch/debtap) to generate a pacman compliant package, but be aware that raspotify is using some debian specific commands (useradd or mkdir in /bin/mkdir and not in /usr/bin/mkdir and so on), so you have to do soma manual work to get it running

Lerentis, 2018-11-11:
Short status update: the approach with debtap works, but you have to modify the service file and move it to /etc/systemd/system/raspotify.service 

```
[Unit]
Description=Raspotify
After=network.target

[Service]
User=raspotify
Group=raspotify
Restart=always
RestartSec=10
PermissionsStartOnly=true
ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify ; /bin/chown raspotify:raspotify /var/cache/raspotify
Environment="DEVICE_NAME=raspotify (%H)"
Environment="BITRATE=160"
Environment="CACHE_ARGS=--disable-audio-cache"
Environment="VOLUME_ARGS=--enable-volume-normalisation --linear-volume --initial-volume=100"
Environment="BACKEND_ARGS=--backend alsa"
EnvironmentFile=-/etc/default/raspotify
ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS

[Install]
WantedBy=multi-user.target
```

also make sure that you have 

* aur/adduser-deb
* extra/alsa-utils

installed

the last point to make it work (and the one that takes the most time) is compiling librespot, but there is a aur package that works for that:

* aur/librespot-git

as raspotify is shipping a compiled version of librespot that does not work with arch arm, you have to install this package with --force to overwrite the other version.
After that configure `/etc/default/raspotify` to your liking and start/enable the systemd service 

Hope that saves you some time ;D

dusatvoj, 2021-11-17:
Not solved.

JasonLG1979, 2021-11-17:
@dusatvoj it's not going to be solved. This project packages librespot in a .deb for Raspberry Pi OS. If you want an Arch package you'll have to roll your own.

Closing this issue because it's invalid.

closed 2021-11-17 by JasonLG1979