dtcooper/raspotify #146: Start Playback from the cloud

skatospag, 2018-09-12:
Hi, 

I want to use the Spotify Web Player API to start playback on my Raspberry, but I want to execute this commande from the cloud (a web server is calling the Spotify API). 

It doesn't work because my Pi is not visible on Spotify connect if I'm not on the same wifi. 
I thought that it should work if I add my Spotify username & password on /etc/default/raspotify config file.
But apparently not, do you have any idea ?

Thanks a lot!

skatospag, 2018-09-12:
Forget what I said, it works fine from a distant network one you have successfully add your Spotify username & password in the config file.
It's just that you have to use your spotify username and not you account email (this is the mistake I made).

skatospag, 2018-09-17:
up someone?

5AMsan, 2018-09-17:
It seems like you already filled username and password in raspotify conf file. 
I have a similar issue with the one at work, I 'm not sure it's a raspotify issue, it seems to be related to Spotify itself since my device is available and working for days and start disapearing without any modification. Restarting the RPI does not help by the way.

skatospag, 2018-09-17:
Thanks for your answer, yes exactly. If someone have found a solution, do not hesitate :)

skatospag, 2018-09-24:
> Just to follow up this issue, while it was impossible to read from Spotify two days ago, everything is fine today. No RPI restart. I would say my local network is less congested today.

For me nothing has changed :(

5AMsan, 2018-09-24:
Sometimes, restarting raspotify service helps : 
```
sudo systemctl restart raspotify
```

110101011, 2018-10-17:
Might be impossible as 'official' Spotify connect device typically only work if they are in the same network. So I could be something that Spotify is blocking. You could setup and VPN server and connect to it then it would accessable every where in the world.

5AMsan, 2018-10-18:
@110101011 Actually, you don't need to be on the same local netwok since if you add Spotify credentials in Raspotify conf, your device will show up on Spotify App :
[image]

110101011, 2018-10-18:
@5AMsan, I thought Spotify would have blocked that, never checked to see.
If it should be accessable outside of the network, forwarding the right port to the Raspberry Pi should do the trick.

5AMsan, 2018-10-18:
@110101011 I never had to modify any port on the gateway or RPi to get it working, going fine out of the box.

closed 2021-11-24 by github-actions[bot]