dtcooper/raspotify #149: ConnectionReset after one title

dandud100, 2018-09-24:
Sep 24 17:30:31 raspberrypi librespot[782]: ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset~

If i just let music play over raspotify, and minimize the spotify app, it just play the actual song to the end and get an ConnectionReset.

Uglendertog, 2018-10-30:
I have the same problem and gets the same error code

. As soon as i minimize the spotify app on my telephone it just plays the song to the end.

Hammit, 2018-12-21:
I too have this issue. I'm using the Ubuntu Linux Spotify Client, not mobile though. Happens after a few songs. Happens when using Android mobile client too.

    Dec 21 10:34:20 music librespot[506]: ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }

lamoe02, 2019-01-28:
same here unfortunately... :(

sfai05, 2019-09-15:
Got this issue after I update the config of BITRATE to 320
Comment out the config then all works as normal

closed 2021-11-24 by github-actions[bot]