dtcooper/raspotify #150: Next, previous, play and pause commands?

e-3-s, 2018-09-29:
Is there a way to to have next, previous, play and pause commands?

vorbiss, 2018-10-08:
same request. I'd like to change track with two buttons.

see my project : [https://forum.hardware.fr/hfr/OverclockingCoolingModding/Divers-8/raspberry-meuble-connecte-sujet_285963_1.htm](url)

110101011, 2018-10-17:
As Spotify connect which raspotify uses is a rather dumb player I don't think it an easy job to implement playlist interaction. It receives commands from another device and follows them.

e-3-s, 2018-10-19:
Hmmmm too bad :(

janekszynal, 2019-01-05:
It's very much possible to do it with Spotify's Web API. I already have part of that in my code, happy to share. If there is enough interest I could clean stuff up and publish a repo (something like raspotify-controls). Would be helpful to know how you guys would like to use them (physical buttons? mobile app? cronjob?)

Briefkastenoma, 2019-02-12:
> It's very much possible to do it with Spotify's Web API. I already have part of that in my code, happy to share. If there is enough interest I could clean stuff up and publish a repo (something like [...]

Hey janekszynal, i would love to control raspotify via Spotifys Web API. Do you have this already running? Using the GPIO with Physical buttons would be the best way to do it.

janekszynal, 2019-02-12:
yes, I have this running. The only things it's currently able to do is authorization and playing a given track (I use it for an alarm clock). But once you're authorized you can do anything can think of, the API is great.

All that I currenly use is in [this file in my repo](https://github.com/janekszynal/spoti-alarm-server/blob/master/webapp/app.py). Unfortunatly, as the README says, it's not exactly share-ready, pretty undocumented and authorization in particular might be confusing. If you wanna do it, but run into problems you can reach out, my gmail is `jan.szynal`. I also wanted to have physical play/pause/next, so we can team up for that :)

Blyzz616, 2020-09-02:
Hi there @janek 
I'm busy building an epaper alarm clock using the raspberry pi zero w.
I've just gotten spotify playing via the mini HDMI out and would love to have an "off" and "sleep" button  integration.
To accomplish this, I'd love to see your code - but unfortunately, the link that you included in your post seems to be dead. :-(

janek, 2020-09-12:
Hi @Blyzz616! Fun to run into someone working on a similar thing. I fixed the link in my previous comment (it was broken because I changed my GitHub username since posting it, sorry). Feel free to reach out to me via email as described or via issues in my repo, as the code there might not be immediately clear (or the best way of doing it).

closed 2021-11-24 by github-actions[bot]