dtcooper/raspotify #152: Unclear on how to disable volume normalisation

ja14000, 2018-10-09:
I am unclear on how to disable the volume normalisation feature without affecting any other of the default settings. 

From what I understand volume normalisation is enabled by default and you must change some arguments to disable it. 

When looking at all of the arguments available there is one available to enable audio normalisation but there doesn't seem to be an argument to disable it.

What I have done is uncommented the VOLUME-ARGS line and removed the' --enable-volume-normalisation' flag.

Is this the correct way to disable volume normalisation?

Thanks

110101011, 2018-10-17:
Looking at librespot remove that line of code should be sufficient to disable volume normalisation

ja14000, 2018-10-18:
Thanks for the response, glad to have confirmation. Please could you point me at the librespot documentation? I'd be interested in reading some more about how this works.

Many thanks

110101011, 2018-10-18:
Documentation for librespot:

https://github.com/librespot-org/librespot/blob/master/README.md

vguttmann, 2020-07-17:
If your issue is resolved, you should consider closing it @ja14000

closed 2020-07-17 by ja14000