dtcooper/raspotify #154: Is raspotify intended to work with pulseaudio?

galwayhooker, 2018-10-19:
I don't know much about audio architecture. my recent pi image is using pulseaudio for everything. is there a way to make raspotify compatible? or should I be looking in librespot?

I am assuming asla-backend and pulseaudio are two different implementations, and ALSA picks whichever is specified? and because raspotify and/or librespot defaults to alsa-backend, it connects and plays, but has no sound since I only have pulseaudio.

Sorry I'm pretty lost in the audio world and its incompatibilities.

mfauvain, 2018-11-28:
Had the same issue where raspotify would throw error with pulseaudio dbus etc..
what I did (with alsa backend):
cd /var/cache/raspotify
sudo mkdir .pulse
sudo sh -c 'echo "default-server = 127.0.0.1" > .pulse/client.conf'
sudo chown -R raspotify:raspotify .pulse

make sure to start raspotify after pulseaudio and that /etc/asound.rc has default to pulse ie :
pcm.!default {
  type pulse
}

ctl.!default {
  type pulse
}

it worked for me, now seeing raspotify in pavucontrol.

andrewtvuong, 2018-12-11:
@mfauvain Unfortunately this did not work. raspotify doesn't seem to work with pulseaudio as default

Lipown, 2020-01-16:
The above post work if below which I copied from Mopidy guide is applied:
```
First, configure PulseAudio to accept sound over TCP from localhost by uncommenting or adding the TCP module to /etc/pulse/default.pa or $XDG_CONFIG_HOME/pulse/default.pa (typically ~/.config/pulse/default.pa):

### Network access (may be configured with paprefs, so leave this commented
### here if you plan to use paprefs)
#load-module module-esound-protocol-tcp
load-module module-native-protocol-tcp auth-ip-acl=127.0.0.1
#load-module module-zeroconf-publish
```

coxtor, 2020-01-25:
@mfauvain  awesome thank you!

digixmax, 2021-03-26:
I am able to set this up on my Pi 3 running Raspian Buster, but not on a Pi 3 running Lubuntu 20.04.2: raspotify output seems stuck on the alsa's default output card regardless of pulseaudio's currently selected output card.

I would appreciate any thought/pointer.

TIA.

JasonLG1979, 2021-11-17:
Raspotify is primarily intended to be used in a headless enviroment (Raspberry Pi OS lite) For desktop versions of Raspberry Pi OS (and/or systems with PulseAudio installed) [spotifyd](https://spotifyd.github.io/spotifyd/installation/Raspberry-Pi.html) offers similar functionality, much better PulseAudio compatibility and is a better choice.

closed 2021-11-17 by JasonLG1979