dtcooper/raspotify #159: Shuffle and Repeat are Reset when Selecting a new Song

ndbeals, 2018-11-04:
Hello!

Thanks for taking the time to read this, as the title states, whenever I'm playing a playlist with shuffle turned on, if I select a new song, it turns off shuffle.

This didn't used to be how it behaved, and I'm wondering how do I get it back?

If there's any more information I can provide, I'd be pleased to.

Thanks,

celeryclub, 2018-11-24:
This appears to be an issue in librespot as well - https://github.com/librespot-org/librespot/issues/266.

ndbeals, 2018-11-24:
Yeah @celeryclub , an issue opened by me because this issue sat unanswered for 19 days, I still don't know who exactly the culprit is.

celeryclub, 2018-11-24:
FWIW I can confirm it's a bug in librespot because I'm running it directly without raspotify and I'm seeing the same behavior you described.

ndbeals, 2018-11-24:
Since this issue does affect Raspotify users (via librespot that is), did you want me to leave this issue open until upstream resolves it?

celeryclub, 2018-11-26:
I don't have a strong opinion on the matter. I suppose we could leave it open, then whenever the issue is resolved upstream this one could be closed after librespot is updated here.

faabmaster, 2019-10-15:
I'm also affected by this issue. Has anyone found a solution to this, considering it's been a year since it was first reported?

willstott101, 2019-10-19:
Pinging raspotify users about https://github.com/librespot-org/librespot/pull/386 please have a go at using that branch to see if the behaviour is exactly as you expect.

closed 2021-03-01 by ndbeals