dtcooper/raspotify #161: Any plan to support free users ?

jkx, 2018-11-13:
The new SDK from Spotify allow free users to use Spotify connect as seen this : 
https://lifehacker.com/spotify-users-with-free-accounts-can-now-use-their-phon-1830280112

I tested it w/ my desktop and droid phone, it seems to work fine. (I don't have a free account, but a middle 4$99) account for years. 

Rasposity still need a premium account.

fosterbuster, 2018-11-14:
You might want to read this: https://github.com/librespot-org/librespot/issues/261
rapotify uses librespot for its backend, so the change has to come from there,

dtcooper, 2018-11-14:
Bit of a heated discussion over there on that librespot issue. But feel free to track it and comment on this issue if they decide to support that, I'll update raspotify. Thanks!

closed 2018-11-14 by dtcooper