dtcooper/raspotify #162: Playing a song via terminal command

kurcab, 2018-11-18:
Does anyone know how I can start a song/playlist from a bash command so that I could schedule stuff with cron?

derWeihnachtsmann, 2018-11-18:
I'm unsure at this but maybe you are searching for something like spotipy ?

3579km, 2018-11-30:
try aplay but not shure how it will be useful

janekszynal, 2019-01-05:
hi @kurcab,

I'm building an alarm clock to wake me up with a song from raspotify and ran into this issue as well (if you're looking to schedule stuff with cron, kind of sounds like your project might be similar).

The bash command would then look something like this:

```curl -X "PUT" "https://api.spotify.com/v1/me/player/play" --data "{\"context_uri\":\"spotify:album:5uiLjgmdPV4dgamvmC64Oq\",\"offset\":{\"position\":5},\"position_ms\":0}" -H "Accept: application/json" -H "Content-Type: application/json" -H "Authorization: Bearer BQDE3bveuC5-0dFy93uvqMGsM5nkwWlhNQbudi7TtjquB58E_cSedGHwYnG7L5Oc76Yvfh11rFrh9eZ5Ln3Y0Otu4VUdRHjU6AjfVoAPIHiv4lBhahVqdz5r1ZZb0e2DDpShAcT0fnAnYeBvhYm1HsUI0Gw_PWaMfCyWiL9S3SA"```

where `spotify:album:5uiLjgmdPV4dgamvmC64Oq` links to the album you want to play (can also play a track with a slightly different syntax) 
and `BQDE3bveuC5(...)` is an Oauth authorisation token you will have to generate from Spotify.

You should start here:
https://developer.spotify.com/documentation/web-api/quick-start/

Let me know if you need any help!

hiyazakite, 2019-03-21:
You can use Spotipy (https://github.com/plamere/spotipy) and create a python script that does this.

closed 2021-11-24 by github-actions[bot]