dtcooper/raspotify #165: Playback stops with Alsa error

meiseGit, 2018-12-04:
I use a Hifiberry DAC+ with a Raspberry Pi 3B and the latest version of raspbian. Since the last weeks I have the following problem. 
After one or two songs the playback stops. Usually it is not possible to start another song until I reboot the pi or restart raspotify. 
"service raspotify status" returns this:

> raspotify.service - Raspotify
>    Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
>    Active: active (running) since Tue 2018-11-27 19:52:49 CET; 6 days ago
>   Process: 529 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
>   Process: 518 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
>  Main PID: 533 (librespot)
>    CGroup: /system.slice/raspotify.service
>            └─533 /usr/bin/librespot --name raspotify (raspberrypi) --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100
> 
> Nov 27 19:56:11 raspberrypi librespot[533]: INFO:librespot_core::session: Authenticated as "XXXXXXXXXXXX" !
> Nov 27 19:56:11 raspberrypi librespot[533]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
> Nov 27 19:56:11 raspberrypi librespot[533]: INFO:librespot_core::session: Country: "DE"
> Nov 27 19:56:11 raspberrypi librespot[533]: INFO:librespot_playback::player: Loading track "Una storia importante" with Spotify URI "spotify:track:61VptdUUGBeSFPGnJD5z8p"
> Nov 27 19:56:17 raspberrypi librespot[533]: INFO:librespot_playback::player: Track "Una storia importante" loaded
> Nov 27 19:56:17 raspberrypi librespot[533]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open -16
> Nov 27 19:56:17 raspberrypi librespot[533]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
> Nov 27 19:56:17 raspberrypi librespot[533]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open -16
> Nov 27 19:56:17 raspberrypi librespot[533]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed

Has anybody an idea what the problem is? Im happy to provide more information if necessary.

kunaakos, 2018-12-10:
I have the same setup and am dealing with the same issue.

warro, 2019-02-20:
Same setup and problem here.

lscobe16, 2019-08-20:
Did you find any solution to this?
Because I'm having the same issue now with the same error message as above except sometimes i can even hear about 10 tracks before the error ocours.
My setup is a Raspberry Pi Zero (not W) with a PhatDAC ontop of it.

closed 2021-11-24 by github-actions[bot]