dtcooper/raspotify #166: connection reset after x songs

trackersos, 2018-12-10:
hello everybody,
the connection of librespot gets lost after 3 or 4 songs. I've already read a lot of this issue, but I couldn't found a solution.
I already checked:
- no running out of space or memory
- no pulseaudio installed
- fresh installation of raspbian 
- running librespot command within a screen session
- no disconnection of my internet

My log says something about code 104. There are as well reported issues about that but restarting the whole service (even automatically) is no solution. I must choose Raspotify after 3 or 4 songs again.

It would be very nice, if somebody can help me.

trackersos

`INFO:librespot_playback::player: Track "Don't Mess With Mister T - Live" loaded
ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
thread 'main' panicked at 'Spirc shut down unexpectedly', src/main.rs:462:25
stack backtrace:
   0:   0x94cd07 - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h63a67ac893ab31ec
                       at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
   1:   0x93bc4b - std::sys_common::backtrace::print::ha8f26509b9ff262c
                       at libstd/sys_common/backtrace.rs:71
                       at libstd/sys_common/backtrace.rs:59
   2:   0x9394c3 - std::panicking::default_hook::{{closure}}::hd7e08be4e57c3548
                       at libstd/panicking.rs:211
   3:   0x939123 - std::panicking::default_hook::h216d888df769ef77
                       at libstd/panicking.rs:227
   4:   0x939beb - std::panicking::rust_panic_with_hook::h7d39a7695d8da98f
                       at libstd/panicking.rs:477
   5:   0x4d3de3 - std::panicking::begin_panic::h624ea99f6c8452d9
   6:   0x4ef3b3 - <librespot::Main as futures::future::Future>::poll::h3db72e6cc182213f
   7:   0x4c9bdf - futures::task_impl::std::set::h7c382029013f638f
   8:   0x4e7c03 - <futures::future::lazy::Lazy<F, R> as futures::future::Future>::poll::h339c16cdcdc8d2bf
   9:   0x4c98cf - futures::task_impl::std::set::h035c704851db453e
  10:   0x4d095b - <std::thread::local::LocalKey<T>>::with::h1726d6c9dbd47ab3
  11:   0x4e6e3f - <tokio::executor::current_thread::Entered<'a, P>>::block_on::h11c540a03761c4aa
  12:   0x4d0f17 - <std::thread::local::LocalKey<T>>::with::h5cfa747dd1304813
  13:   0x4d0a87 - <std::thread::local::LocalKey<T>>::with::h31d282700fc02d1b
  14:   0x4d0cc3 - <std::thread::local::LocalKey<T>>::with::h41159fec93afe179
  15:   0x4d1ba3 - <scoped_tls::ScopedKey<T>>::set::h9677cf0748583291
  16:   0x4e38ff - tokio_core::reactor::Core::run::hf1b7213e939ae7d9
  17:   0x4f1227 - librespot::main::hffab8cbb1bfad5ce
  18:   0x4e41ff - std::rt::lang_start::{{closure}}::h7734a4273cdf8ce1
  19:   0x93960f - std::panicking::try::do_call::he4497885895661a8
                       at libstd/rt.rs:59
                       at libstd/panicking.rs:310
  20:   0x959f8b - __rust_maybe_catch_panic
                       at libpanic_unwind/lib.rs:103
  21:   0x9386a3 - std::panic::catch_unwind::hdd9654d0e8ee1176
                       at libstd/panicking.rs:289
                       at libstd/panic.rs:392
  22:   0x9354ab - std::rt::lang_start_internal::h4ec5eae1ccf41136
                       at libstd/rt.rs:58
  23:   0x4f1e8f - main
  24: 0x76d40677 - __libc_start_main`

fosterbuster, 2018-12-24:
It seems there was a bug, where librespot would try and seek over EOF. It has been fixed, and seems to work for me. So just compile raspotify yourself, and you'll get the fix.

Blechzwerg, 2019-02-12:
Hmmm....

I'm experiencing the same(?) issue. 
What do I need to do to get such a complete debug log?  My stops at the very beginning with the 104 message...

You say, this might be fixed, I'd just need to re-compile... 

How did I do that on a Raspi?  Nothing of whatever I found regarding Rust did work (got recognized by my Raspi...).  Can you link me to a tutorial about the build?

;-)
E.

P.S. I installed raspotify on my Raspi

Brucki, 2019-03-04:
Have you tried to disable any battery saving modes?
Had the same problem with my HUAWEI phone, it's working now since i disabled optimisations for the Spotify app.

Settings -> Battery -> Launch -> Spotify ... Manage manually & allow all

Blechzwerg, 2019-03-04:
For me it doesn't even has to do with Android app or sth.  
It also happens if I control by Win client or from the web browser.
;-)
b.

tiramon, 2019-05-01:
Still crashing with  newest release v0.13.1

`May  1 11:40:30 raspberrypi librespot[3332]: ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
May  1 11:40:30 raspberrypi librespot[3332]: thread 'main' panicked at 'Spirc shut down unexpectedly', src/main.rs:497:25
May  1 11:40:30 raspberrypi librespot[3332]: stack backtrace:
May  1 11:40:30 raspberrypi librespot[3332]:    0:   0x9d4fa7 - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h71a4d52fb38cd7de
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:39
May  1 11:40:30 raspberrypi librespot[3332]:    1:   0x9d066f - std::sys_common::backtrace::_print::hce8fc7b28ee67394
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/sys_common/backtrace.rs:70
May  1 11:40:30 raspberrypi librespot[3332]:    2:   0x9d3193 - std::panicking::default_hook::{{closure}}::hdaad4cfa8d5b3133
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/sys_common/backtrace.rs:58
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/panicking.rs:200
May  1 11:40:30 raspberrypi librespot[3332]:    3:   0x9d2dcf - std::panicking::default_hook::hfb698835042c0732
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/panicking.rs:215
May  1 11:40:30 raspberrypi librespot[3332]:    4:   0x9d38bb - std::panicking::rust_panic_with_hook::h25201d3872ea325e
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/panicking.rs:478
May  1 11:40:30 raspberrypi librespot[3332]:    5:   0x4f51db - std::panicking::begin_panic::hd098aa9360cc2a50
May  1 11:40:30 raspberrypi librespot[3332]:    6:   0x50bc5f - <librespot::Main as futures::future::Future>::poll::h3955ff8efce85de2
May  1 11:40:30 raspberrypi librespot[3332]:    7:   0x4ec73f - futures::task_impl::std::set::h04ef672383e9a200
May  1 11:40:30 raspberrypi librespot[3332]:    8:   0x514fb7 - <futures::future::lazy::Lazy<F, R> as futures::future::Future>::poll::ha33375dd94f167fe
May  1 11:40:30 raspberrypi librespot[3332]:    9:   0x4eca4f - futures::task_impl::std::set::hb612a7615f6fd096
May  1 11:40:30 raspberrypi librespot[3332]:   10:   0x4f56b7 - <std::thread::local::LocalKey<T>>::with::hcc9f13158a697b9a
May  1 11:40:30 raspberrypi librespot[3332]:   11:   0x4f4fe3 - <scoped_tls::ScopedKey<T>>::set::h76dbe7a12395256f
May  1 11:40:30 raspberrypi librespot[3332]:   12:   0x50f197 - tokio_core::reactor::Core::run::h12e3b01e0232aff0
May  1 11:40:30 raspberrypi librespot[3332]:   13:   0x50dfb3 - librespot::main::h55151ad450023eb3
May  1 11:40:30 raspberrypi librespot[3332]:   14:   0x4ebca3 - std::rt::lang_start::{{closure}}::h5d8a9a9c38c99203
May  1 11:40:30 raspberrypi librespot[3332]:   15:   0x9d32e3 - std::panicking::try::do_call::h25e833b9f76f9c1b
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/rt.rs:49
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/panicking.rs:297
May  1 11:40:30 raspberrypi librespot[3332]:   16:   0x9db1c3 - __rust_maybe_catch_panic
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libpanic_unwind/lib.rs:87
May  1 11:40:30 raspberrypi librespot[3332]:   17:   0x9d3d4b - std::rt::lang_start_internal::haee47fdaef2afaf3
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/panicking.rs:276
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/panic.rs:388
May  1 11:40:30 raspberrypi librespot[3332]:                        at src/libstd/rt.rs:48
May  1 11:40:30 raspberrypi librespot[3332]:   18:   0x50edcf - main
May  1 11:40:30 raspberrypi librespot[3332]:   19: 0x76dd9677 - __libc_start_main
May  1 11:40:30 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
May  1 11:40:30 raspberrypi systemd[1]: raspotify.service: Unit entered failed state.
May  1 11:40:30 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
`

oridanus, 2019-06-05:
any update on this?

david-knigge, 2020-03-02:
```
Mar 02 21:54:48 rpi-ams-01 librespot[2528]: [2020-03-02T21:54:48Z INFO  librespot_playback::player] <Feel First Life> loaded                                                                                                                    
Mar 02 21:55:03 rpi-ams-01 librespot[2528]: [2020-03-02T21:55:03Z ERROR librespot_core::session] Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }                                                                   
Mar 02 21:55:03 rpi-ams-01 librespot[2528]: [2020-03-02T21:55:03Z WARN  librespot_audio::fetch] Error from channel for data receiver for range 1639896 (+16384).   
```

Seems like a similar error still, any updates?

jkwcht, 2020-04-03:
[https://github.com/dtcooper/raspotify/issues/112#issuecomment-608361253](https://github.com/dtcooper/raspotify/issues/112#issuecomment-608361253)

closed 2021-11-24 by github-actions[bot]