dtcooper/raspotify #167: Support read-only Raspbian Stretch Lite

kyleaffolder, 2018-12-15:
Hi!

Great program and appreciate all of your hard work on this! I have been using it with no issues the last few weeks. I wanted to reach out though as I would like to get raspotify running using a Raspberry Pi in read-only mode.

Using [this script](https://learn.adafruit.com/read-only-raspberry-pi/overview) and method on the [Adafruit](http://adafruit.com) website to set a working image to read-only, I have successfully set up Raspbian Stretch (Lite) into read-only mode. Before this conversion, raspotify worked as expected. After running this script and SSH'ing into the system, I see the following error while running `sudo systemctl status raspotify`:

```
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2018-12-15 16:34:43 EST; 5s ago
  Process: 2013 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=1/FAILURE)
  Process: 2010 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)

Dec 15 16:34:43 sangean-wr16-radio systemd[1]: Failed to start Raspotify.
Dec 15 16:34:43 sangean-wr16-radio systemd[1]: raspotify.service: Unit entered failed state.
Dec 15 16:34:43 sangean-wr16-radio systemd[1]: raspotify.service: Failed with result 'exit-code'.
```

I would really like to get this system running as read-only, as it helps with unexpected shutdowns, so any help is appreciated.

Much thanks in advance!

-Kyle

PS. I am using a phatDAC hat, which shouldn't matter and works fine as before.

dtcooper, 2019-01-07:
Ah, you might try to remove the `ExecStartPre` lines with `chown` and `mkdir` from your systemd service file. Let us know if that works!

closed 2019-01-07 by dtcooper

Ja83La, 2020-04-10:
Commenting out (#) the line beginning with ExecStartPre  /lib/systemd/system/raspotify.service works fine!

JimmyUX, 2021-05-07:
Hi, I am also having this problem. Even after I commented out the lines starting with ```ExecStartPre```
I am able to see the Spotify Connect device, but if I try to play to it, Spotify just skips through the songs in the current playlist.