dtcooper/raspotify #171: Possible to see current song?

thecompix, 2018-12-28:
Hey Guys,

is it possible to see via command the current playing song with artist, title and track time?
because i want to put this output to a webview with js variables.

thaaanks guys
the compix.

janekszynal, 2019-01-05:
hey @thecompix,

you might want to check out Spotify's Web API. It's well done and opens up a whole world of possibilities if you're hacking around. I'm using it with raspotify and I'm very happy. Start here:
https://developer.spotify.com/documentation/web-api/quick-start/ 

The endpoint that will return the current song playing is documented here: 
https://developer.spotify.com/documentation/web-api/reference/player/get-the-users-currently-playing-track/

And you can try it out here:
https://developer.spotify.com/console/get-users-currently-playing-track/

If you run into problems let me know here and I'll try to help out.

fixmycode, 2019-02-27:
I work on a lab and we have a RPi with raspotify connected to some old speakers. The device is shared among all the lab workers, with many different spotify accounts, and the web API depends on a single account to show the information. Being able to read the current track using a frontend program would be nice, something like `raspotify --now-playing`, that way I could show what's playing in the lab screens...

I'd label this issue as a feature request.

l3d00m, 2019-04-02:
Related? https://github.com/librespot-org/librespot/issues/185

Ich-Eben, 2019-07-01:
Hey,
the --onevent option from librespot works great btw. So we don’t really need to implement this in raspotify. Just add:
`OPTIONS="--onevent 'node /usr/local/bin/myEventGateway/main.js'"`
To your raspotify config file and put this script in /usr/local/bin/myEventGateway/main.js (for example):
```
var mqtt = require('mqtt');
var client  = mqtt.connect('mqtt://localhost');

client.on('connect', function() {
        client.publish('raspotify/playEvent', process.env.PLAYER_EVENT);
        client.publish('raspotify/trackId', process.env.TRACK_ID);
        client.publish('raspotify/oldTrackId', process.env.OLD_TRACK_ID);
        client.end();
});
```
This will publish the events send from librespot to an mqtt server. Than you can grab it from there from pretty everywhere. Of course you'll still have to use the WebAPI to get the song name from the track id, but you don’t need an account for this as far as I know.
Don’t forget to install mosquito and nodejs on your pi to use this. You’ll also require the node module mqtt (npm install mqtt).
hf

closed 2021-11-24 by github-actions[bot]